lnis-uofu/OpenFPGA #795: Testbench gives multiples 1'bx when use high number of inputs/outputs

Baungarten-CINVESTAV, 2022-09-18:
Hello every one,

As the title says i had a couple of erros when i used mutiples i/o. Trying to find the mistake, i simplified my modules on a couple of adders:

The module [adder_128_with_vector ](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/benchmarks/My_bench_Mark/adder_128/adder_128_with_vector.v)is a simple add of two numbers of 128 bits, the output has 129 bits and the i/o are defined as a unitary vector, e.g. input [127:0] i_num_1. 
[image]

The module [adder_128_without_vector ](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/benchmarks/My_bench_Mark/adder_128/adder_128_without_vector.v)is a simple add of two numbers of 128 bits, the output has 129 bits and each i/o are defined 1 by 1, I thought the problem might be the definition of i/o.  
[image]

the module [adder_128SSE ](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/benchmarks/My_bench_Mark/adder_128/adder_128SSE.v)perform 4 adds of two numbres of 32 bits but it received just 1 input of 128 bits. 
[image]

It can be seen in the previous images that each of the above testbenches has an error or unknow value.

I want to know what is wrong in my project and how i can solve my problem?

share my [openfpga_flow ](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test)folder

[https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test)

Thanks in advance

Baungarten-CINVESTAV, 2022-09-19:
Sorry, I forgot to add this information.
Those testbench was run with the task [preconfigured_testbench_without_self_checking](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/tree/master/tasks/basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/config)

tangxifan, 2022-09-23:
@Baungarten-CINVESTAV Can you check if the post-synthesis netlists from Yosys is correct or not? This will help us to spot the bug.

Baungarten-CINVESTAV, 2022-09-23:
Hello @tangxifan 

With the post-synthesis netlists from Yosys file are you talking about [xxx_ace_out.blif](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/tasks/basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/run001/k6_frac_N10_tileable_40nm/adder_128_with_vector/MIN_ROUTE_CHAN_WIDTH/adder_128_with_vector_ace_out.blif),[ xxx_yosys_out.blif](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/tasks/basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/run001/k6_frac_N10_tileable_40nm/adder_128_with_vector/MIN_ROUTE_CHAN_WIDTH/adder_128_with_vector_yosys_out.blif), [xxx.blif](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/tasks/basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/run001/k6_frac_N10_tileable_40nm/adder_128_with_vector/MIN_ROUTE_CHAN_WIDTH/adder_128_with_vector.blif) , [xxx.net](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/tasks/basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/run001/k6_frac_N10_tileable_40nm/adder_128_with_vector/MIN_ROUTE_CHAN_WIDTH/adder_128_with_vector.net) or  [fabric_netlist.v](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/tasks/basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/run001/k6_frac_N10_tileable_40nm/adder_128_with_vector/MIN_ROUTE_CHAN_WIDTH/SRC/fabric_netlists.v) file?

Baungarten-CINVESTAV, 2022-10-09:
> @Baungarten-CINVESTAV Can you check if the post-synthesis netlists from Yosys is correct or not? This will help us to spot the bug.

Hello,

I have checked all the .blif files and their boolean equations are correct, to test the blif file, I developed a matlab script that takes the blif file and gets the SOP boolean equation for each output, then I compared the result of the boolean equation with a golden variable that stores a 128 bit sum. I did the same with the 3 blif files and everything works fine.

I also developed a testbench of the [adder_128_with_vector_output_verilog.v](https://github.com/Baungarten-CINVESTAV/OpenFPGA_addr_test/blob/master/tasks/basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/run001/k6_frac_N10_tileable_40nm/adder_128_with_vector/MIN_ROUTE_CHAN_WIDTH/adder_128_with_vector_output_verilog.v) (The verilog file made by yosys based on the .blif file) and the result was the same the module develops well the sum. 
[image]

What do you think is the problem?

Baungarten-CINVESTAV, 2022-11-08:
This issues was solve in #880, Summary, icarous verilog has some flaws.

closed 2022-11-08 by Baungarten-CINVESTAV