lnis-uofu/OpenFPGA #797: issue on reset

zliu1Charlotte, 2022-09-18:
Hi,

I am trying the micro_benchmark/counters/counter_8bit_async_reset/counter.v design on this task.conf setting:

openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/preconfigured_testbench_without_self_checking_example _script.openfpga
openfpga_arch_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_arch/k6_frac_N10_tree_mux_40nm_openfpga.xml
openfpga_sim_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml
openfpga_pin_constraints_file = ${PATH:OPENFPGA_PATH}/openfpga_flow/tasks/basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/config/pin_constraints.xml
arch0=${PATH:OPENFPGA_PATH}/openfpga_flow/vpr_arch/k6_frac_N10_tileable_40nm.xml
bench1=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/counters/counter_8bit_async_reset/counter.v
bench_read_verilog_options_common = -nolatches

But the output of the counter stays as 'x' when clock and reset changes. 
I know we need to set the constrains:
<set_io pin="reset[0]" net="reset"/>

but this still did not change the 'x' output.

My first question is: how do I fix this? 

Then, I tried this in the task.conf:
openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/example_without_ace_script.openfpga
openfpga_arch_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_arch/k4_frac_N4_fracff_40nm_cc_openfpga.xml
openfpga_sim_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_simulation_settings/fixed_sim_openfpga.xml
arch0=${PATH:OPENFPGA_PATH}/openfpga_flow/vpr_arch/k4_frac_N4_tileable_fracff_40nm.xml
bench1=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/counters/counter_8bit_async_reset/counter.v
bench_read_verilog_options_common = -nolatches
bench1_openfpga_pin_constraints_file = ${PATH:OPENFPGA_PATH}/openfpga_flow/tasks/basic_tests/k4_series/k4n4_fracff/config/pin_constraints_reset.xml

this now gives me the correct output on the counter, so I tried this on my own design as well (a counter follow by a sbox). it also shows correct output but when i do vvp compiled_counterSBox:
VCD info: dumpfile counterSBox_formal.vcd opened for output.
Simulation start
Mismatch on out_0__gfpga at time =               2.00ns
Mismatch on out_1__gfpga at time =               2.00ns
Mismatch on out_5__gfpga at time =               2.00ns
Mismatch on out_6__gfpga at time =               2.00ns
Simulation Failed with           4 error(s)

My second question is: Is there way to fix this time delay? 

Thank you @tangxifan

zliu1Charlotte, 2022-09-20:
I have resolved the first two questions. 
My only question left is that when I port my netlists into vivado, gfpga outputs are all "z". 
in the gtkwave I see the correct gfpga but do not understand why in vivado all gfpga are all 'z'. 
Thanks.

zliu1Charlotte, 2022-09-22:
I am able to resolve this issues, if anyone else have the same questions, feel free to leave a message to me.

closed 2022-09-22 by zliu1Charlotte

Baungarten-CINVESTAV, 2022-10-12:
> I am able to resolve this issues, if anyone else have the same questions, feel free to leave a message to me.

Hello,

Please could you tell me how you fixed the "x" values during simulation?

Your answer could help me to solve my issue [#795](https://github.com/lnis-uofu/OpenFPGA/issues/795)

zliu1Charlotte, 2022-10-21:
The issue is on the user_defined_templates.v. it is empty. 
You need the corresponding dff.v and gpio.v