lnis-uofu/OpenFPGA #798: The best CLB architecture features

msaideroglu, 2022-09-19:
What input number should be given when defining vpr_arch file? 
`<tile name="clb" area="53894">
      <equivalent_sites>
        <site pb_type="clb"/>
      </equivalent_sites>
      <input name="I" num_pins="40" equivalent="full"/>
      <output name="O" num_pins="10" equivalent="none"/>
      <clock name="clk" num_pins="1"/>
      <fc in_type="frac" in_val="0.15" out_type="frac" out_val="0.10"/>
      <pinlocations pattern="spread"/>
    </tile>`
For instance why input number 40 is given in k6_frac_N10 architecture definition? I saw an expression in @tangxifan 's EPFL thesis paper that `In modern FPGAs,
the best CLB architecture is typically featured by K = 6, N = 10 and I = K (N + 1)/2 = 33` . Where can I find this architectural informations? Are all input numbers supported and can be built by OpenFPGA?

tangxifan, 2022-09-23:
@msaideroglu 

The classical formula is applied when there are no hard adders in the CLB. It can promise your ~98% utilization of CLB resources, as stated in Prof. Vaughn Betz's thesis.

However, when you have hard adders in CLB, you need more independent inputs, if you want to implement a full-adder using a CLB. Take the example of k6_frac_N10 architecture, there are 20-bit hard adders, which require at least 40 inputs from CLBs to drive them (20-bit for input ``a`` while 20-bit for input ``b``).  That is why we level up the number of inputs.

closed 2022-09-23 by tangxifan

msaideroglu, 2022-09-23:
Thanks @tangxifan. Clearly understood.

msaideroglu, 2022-09-30:
Hi @tangxifan again. I checked **k4_frac_N4_adder_chain.xml** arch file and the inout definition is like 
`<input name="I" num_pins="12" equivalent="full"/>
<input name="cin" num_pins="1"/>
<output name="O" num_pins="8" equivalent="none"/>
<output name="cout" num_pins="1"/>`

If there are 8-bit hard adders don't we need 16-bit inputs instead of 12? Which point am I missing?