lnis-uofu/OpenFPGA #799: Circuit requires a channel width above 1000

msaideroglu, 2022-09-19:
When I use **k6_frac_N10_tileable_adder_chain_40nm.xml** instead of **k6_frac_N10_adder_chain_40nm.xml** to synthesize a counter design with auto_layout feature, it gives `This circuit requires a channel width above 1000, probably is not going to route.` error. Although there is no layout size restriction, why vpr can not route the design with tileable architecture, but can route successfully with non-tileable architecture? @tangxifan

tangxifan, 2022-09-23:
@msaideroglu This is a known bug in tileable routing resource graph. If you try a bigger array size, e.g., 2x2, it should work fine then. Since most FPGA fabrics are way bigger than 2x2, we do not see such problems in practice often. Therefore, it is not so urgent for use to fix.

See details in https://github.com/lnis-uofu/OpenFPGA/issues/71

closed 2022-09-23 by tangxifan

msaideroglu, 2022-09-23:
Thanks @tangxifan for clear explanation.