lnis-uofu/OpenFPGA #801: openfpga visualization

narutozxp, 2022-09-21:
The VPR in openfpga does not seem to support visualization, so can we use the original VPR in the VTR to generate the placement and routing results, as well as the routing resource map, and then use openfpga to generate the corresponding HDL code?

tangxifan, 2022-09-23:
@narutozxp Yes. You can use the original VPR to visualize the results. The HDL codes of FPGA fabrics generated by FPGA-Verilog are consistent (unless you change your architecture description).

We will bring the visualization back to OpenFPGA shell.

narutozxp, 2022-09-26:
The problem now is that I first generate the RTL file of the FPGA through openfpga, and the layout and routing results of the corresponding benchmark. But we found that vpr's performance evaluation results are far more optimistic than DC, so we want to locate the problem by observing the routing results of simple circuits, but openfpga cannot be visualized, so we use VTR to achieve our purpose, but this method also has a problem, Because we want VTR to visualize our previous place and route results (instead of re-running the place and route process), we use the --analysis method, but since the architecture files used by openfpga and VTR are not exactly the same, and --analysis will Check the hash value of the architecture file, so even modifying the architecture file will not achieve our purpose