lnis-uofu/OpenFPGA #860: full testbench on adder

zliu1Charlotte, 2022-10-25:
Hi, I am using the task file from /OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/config/
and I added a benchmark:
bench3=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/adder/adder_8/adder_8.v
I want to use the full testbench.
The openfpga shell script is working fine but it froze at Launching vvp_verification.
So it never generates the vcd file. 
Is there a way to fix this?

tangxifan, 2022-10-25:
Please do not use iverilog when you have complex fabric. Runtime is very long. Recommend commercial HDL simulators, like Modelsim.

closed 2022-10-25 by zliu1Charlotte

Lukemagik, 2024-03-12:
@tangxifan Good morning, I'm encountering a similar issue in various full_testbench instances, such as in the openfpga_flow/fpga_verilog/dsp/single_mode_mult_8x8 task. However, it doesn't seem to be an issue of simulation time, but rather the simulation appears to be stuck at a certain point during the bitstream loading process.

chaitalisathe, 2024-04-09:
> Hi, I am using the task file from /OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/config/ and I added a benchmark: bench3=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/micro_benchmark/adder/adder_8/adder_8.v I want to use the full testbench. The openfpga shell script is working fine but it froze at Launching vvp_verification. So it never generates the vcd file. Is there a way to fix this? I checked the vcd file, the bit_index is at hex 0868, and it somehow froze at hex 0868, but eventually it needs to get to ibit value which is hex 0AF0, but I doubt that it takes such a long time to load the bitstream...

Hello @zliu1Charlotte Did you try Modelsim simulation? Is it working for you?
I tried running Modelsim simulation and I am getting error as- Net type of ' port_name' was not explicitly declared. for almost all ports which aren't defined specifically as wire or reg type.