lnis-uofu/OpenFPGA #880: Simulation errors of icarus verilog vs. commercial HDL simulators

Baungarten-CINVESTAV, 2022-11-07:
> **Describe the bug**
>When developing a 32-bit adder with the k6_frac_N10_adder architecture the ".vcd" file shows unknown values, this compiling with icarous verilog and checking the signal with GTKWave. On the other hand, taking the same verilog files and running the simulation with commercial simulators (QuestaSim) the output has some changes and then the output value is correct.

> **To Reproduce**
> 1- Go to task: basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking/config/task.conf
> 2- Change the OpenFPGA and VPR architecture to:
>> - openfpga_arch_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_arch/k6_frac_N10_adder_chain_40nm_openfpga.xml
>> - arch0=${PATH:OPENFPGA_PATH}/openfpga_flow/vpr_arch/k6_frac_N10_tileable_adder_chain_40nm.xml

> 3- Create a simple 32-bit adder, add it to the benchmarks and choose it as the top module.
>>  e.g:
>>  - [ADD32.txt](https://github.com/lnis-uofu/OpenFPGA/files/9953550/ADD32.txt)
>> - bench0=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks/My_bench/ADD32.v
>> - bench0_top = ADD32

> 4. Execute OpenFPGA task
>> - python3 openfpga_flow/scripts/run_fpga_task.py basic_tests/preconfig_testbench/preconfigured_testbench_without_self_checking --debug --show_thread_logs

> 5. Run the ".vdc" file with GTKwave
> 6. Check the test bench

> **Expected behavior**
> GTKwave displays multiple unknown values
>>  [image]

> Simulating the same Verilog files with QuestaSim the result is:
[image]

> Additional information
The ".vcd" file generated by QuestaSim works fine.
[image]

> **Enviornment (please complete the following information):**

>  - OS: 
< ---    - [x] Ubuntu 20.04.5 LTS --->

>  - Compiler:
< ---    - [x] gcc-9 --->

> - Version:
< ---    - [x] Current master --->

> **Questions section**
> - Why does the icaros verilog give wrong signals?
> - Do multiple changes before the correct result affect the system?
>> [image]

> - Is it correct to think that the FPGA and the system on it are well implemented?

tangxifan, 2022-11-07:
@Baungarten-CINVESTAV It seems a bug of icarus Verilog in my view. Since we have no access to commercial simulators in OpenFPGA CI, we have to use iVerilog for simple and quick verification. I suggest you stay with commercial simulators in your current project.

Baungarten-CINVESTAV, 2022-11-08:
Thank you for your quick response