lnis-uofu/OpenFPGA #885: VPR does not map the BLIF on VTR architecture file

yunuseryilmaz18, 2022-11-09:
> **Describe the bug**
> I have a BLIF and I want to see it is mapped on my VTR architecture file by using VPR. However, when I run VPR, I see `Message: Port 'data_in[1]' on architecture model 'spram64x32' exceeds port width (1 bits)` error. I've changed the bit widths of architecture file and BLIF but the VPR still does not map. What should I do?

> Which part of OpenFPGA is buggy
[ ] Documentation 
[ ] OpenFPGA flow 
[ ] FPGA-Verilog 
[ ] FPGA-Bitstream 
[ ] FPGA-SDC 
[ ] FPGA-SPICE 
[X] VPR 
[ ] Yosys 

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: 874d7fa14d2ed1513701e59f3a3ce61b0941ff70
> 2. Run `$vpr ../arch/k4_frac_N8_tileable_adder_chain_mem1K_dsp8_fracff2edge.xml ../benchmark/dpram_mac_dff/dpram_mac_dff_yosys_out.blif --disp on --device 6x6`
> 3. See error

> **Expected behavior**
> I want to see that my circuit is mapped on VPR.

> **Screenshots**
> [image]

> **Enviornment (please complete the following information):**

>  - OS: 
> - [X] Ubuntu 20.04

>  - Compiler:
> - [X] gcc-9

> - Version:
> - [X] Current master

> **Additional context**
> I've upload my arch file and BLIF: 
[dpram_mac_dff.zip](https://github.com/lnis-uofu/OpenFPGA/files/9971412/dpram_mac_dff.zip)

tangxifan, 2022-11-09:
@yunuseryilmaz18 I see you have defined two blif model ``spram64x32`` with different port sizes. That may be the root of cause.

[image]

yunuseryilmaz18, 2022-11-10:
@tangxifan Deleted the `blif_model` in the physical mode, now I get this error:
`Error 1: ../arch/k4_frac_N8_tileable_adder_chain_mem1K_dsp8_fracff2edge.xml:1197 Unexpected child 'T_setup' of node 'pb_type'. Expected (possibly) one of: 'input', 'output', 'clock', 'mode', 'power', 'metadata', 'model', 'pb_type', or 'interconnect'.`

tangxifan, 2022-11-17:
@yunuseryilmaz18 For any primitive pb_type, you need a blif model. You may consider to create two blif models in different names.

yunuseryilmaz18, 2022-11-18:
@tangxifan I gave a different name for the physical mode but it still gives the same error, `Message: Port 'data_in[1]' on architecture model 'spram64x32' exceeds port width (1 bits)`

tangxifan, 2022-11-18:
@yunuseryilmaz18 Can you try to remove the ``class=memory`` as well as ``port_class``?

yunuseryilmaz18, 2022-11-21:
@tangxifan Removing `class=memory` and `port_class` helped me to pass the placement stage without errors, but now the routing stage is failing with 
```Warning 376: No routing path for connection to sink_rr 491, retrying with full device bounding box
Cannot route from io[5].inpad[0] (RR node: 5 class: 11 capacity: 1 fan-in: 0 fan-out: 1 SOURCE:5 (1,0)) to memory[0].addr_b[4] (RR node: 491 class: 13 capacity: 1 fan-in: 1 fan-out: 0 SINK:491 (3,6)) -- no possible path
Failed to route connection from 'raddr[4]' to 'dout[0]' for net 'raddr[4]' (#10)
Routing failed for net 10
# Routing took 0,02 seconds (max_rss 60,0 MiB, delta_rss +0,0 MiB)
Circuit is unroutable with a channel width factor of 200.

tangxifan, 2022-11-21:
@yunuseryilmaz18 Please share you log file so that it is easier to spot the bug.

yunuseryilmaz18, 2022-11-22:
@tangxifan I've added the log file.
[vpr_stdout.log](https://github.com/lnis-uofu/OpenFPGA/files/10069574/vpr_stdout.log)

tangxifan, 2022-11-22:
> @tangxifan I've added the log file. [vpr_stdout.log](https://github.com/lnis-uofu/OpenFPGA/files/10069574/vpr_stdout.log)

@yunuseryilmaz18 

I see some pins are reachable reported by VPR. (See LINE363 in your log file)

This could be due to the pin locations of your BRAM tile.
Here are the things that you may try

- Enable ``[through_channel](https://openfpga.readthedocs.io/en/master/manual/arch_lang/addon_vpr_syntax/#cmdoption-arg-through_channel)`` in the tileable options. The ``spread`` style of VPR may cause some pins inaccessible when using the tileable architecture. To check that, you can visualize the VPR architecture and check the pin location of BRAM tiles. Regarding how to visualize the architecture, check #801 

- I suggest you to define the pin location in a specific way. Here is an example. In this case, you may not need the ``through_channel`` to be enabled.

https://github.com/lnis-uofu/OpenFPGA/blob/d6eec67c2ee2b94a475e1702889a9768c3645445/openfpga_flow/vpr_arch/k6_frac_N10_tileable_adder_chain_mem1K_40nm.xml#L185-L207

yunuseryilmaz18, 2022-11-23:
> > @tangxifan I've added the log file. [vpr_stdout.log](https://github.com/lnis-uofu/OpenFPGA/files/10069574/vpr_stdout.log)
> 
> @yunuseryilmaz18
> 
> I see some pins are reachable reported by VPR. (See LINE363 in your log file)
> 
> This could be due to the pin locations of your BRAM tile. Here are the things that you may try
> 
>     * Enable [through_channel](https://openfpga.readthedocs.io/en/master/manual/arch_lang/addon_vpr_syntax/#cmdoption-arg-through_channel) in the tileable options. The `spread` style of VPR may cause some pins inaccessible when using the tileable architecture. To check that, you can visualize the VPR architecture and check the pin location of BRAM tiles. Regarding how to visualize the architecture, check [openfpga visualization #801](https://github.com/lnis-uofu/OpenFPGA/issues/801)
> 
>     * I suggest you to define the pin location in a specific way. Here is an example. In this case, you may not need the `through_channel` to be enabled.
> 
> 
> https://github.com/lnis-uofu/OpenFPGA/blob/d6eec67c2ee2b94a475e1702889a9768c3645445/openfpga_flow/vpr_arch/k6_frac_N10_tileable_adder_chain_mem1K_40nm.xml#L185-L207

@tangxifan Thank you very much, it routes now. Do pin locations affect performance and area, is there any research about it?

tangxifan, 2022-11-23:
@yunuseryilmaz18 You may refer to Prof. Vaugh's book, which discusses about impacts of the pin location on P.P.A.

tangxifan, 2022-11-23:
@yunuseryilmaz18 Close the issue now as the problem is resolved. Feel free to open if you see further issues.

closed 2022-11-23 by tangxifan