lnis-uofu/OpenFPGA #889: Utilization on sb and cb

zliu1Charlotte, 2022-11-11:
Hi, do we know what is the utilization on sb and cb for each benchmark?
I see we have the block utilization posted on clb and io in vpr_stdout.log.
Do we have the same report on sb and cb?
Thanks!

tangxifan, 2022-11-17:
Currently, I do not have the stats. But it should be fairly easy to create. In general, if you are interested in the routing utilization, you can find these numbers by enabling ``--full_stats on`` when running vpr.

zliu1Charlotte, 2022-12-05:
Hi, I add vpr ${VPR_ARCH_FILE} ${VPR_TESTBENCH_BLIF} --clock_modeling route --full_stats on
But I am having trouble finding the utilization of sb and cb. 
[vpr_stdout.log](https://github.com/lnis-uofu/OpenFPGA/files/10158321/vpr_stdout.log)

tangxifan, 2022-12-05:
@zliu1Charlotte VPR only outputs the overall routing utilization. There are no CB/SB utilization reported.

You can check the LINE641-735 in your attached log file.

zliu1Charlotte, 2022-12-05:
Thanks, but the log file is rather confusion. 
Terminologies like `Maximum routing channel utilization:      0.67 at (2,0)`
what is 2.0, and does this mean the routing utilization is 0.67?

zliu1Charlotte, 2022-12-05:
I add `--disp on` and `--save_graphics on` but I am having trouble finding its outputs `vpr_placement.pdf and vpr_routing.pdf`

zliu1Charlotte, 2022-12-05:
How to properly interpret this histogram?
  Routing channel utilization histogram:
  [        1:      inf) 0 ( 0.0%) |
  [      0.9:        1) 0 ( 0.0%) |
  [      0.8:      0.9) 0 ( 0.0%) |
  [      0.7:      0.8) 4 (12.5%) |*************************
  [      0.5:      0.6) 2 ( 6.2%) |************
  [      0.4:      0.5) 6 (18.8%) |*************************************
  [      0.3:      0.4) 4 (12.5%) |*************************
  [      0.2:      0.3) 8 (25.0%) |*************************************************
  [      0.1:      0.2) 0 ( 0.0%) |
  [        0:      0.1) 8 (25.0%) |*************************************************
  Maximum routing channel utilization:      0.67 at (2,0)

thanks.

closed 2022-12-08 by zliu1Charlotte

tangxifan, 2022-12-08:
@zliu1Charlotte

- ``Maximum routing channel utilization: 0.67 at (2,0)`` means that the coordinate (2,0), 67% of the routing tracks in that routing is used for signal routing. I am not pretty clear about the histogram. VTR community should have better understanding. Or you can try to search it in their documentation.

https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#placement-file-format-place

- ``Segment usage by type (index): type utilization`` indicates the utilization rate of different type of wires/routing tracks in a routing channel. In FPGA,  there could be multiple types of wires in a routing channel, each of which can speed up the connections. I suggest you to read Prof. Vaughn's book to understand the basic concepts： https://tspace.library.utoronto.ca/handle/1807/12758

zliu1Charlotte, 2022-12-08:
@tangxifan  Thank you