lnis-uofu/OpenFPGA #891: Failed to execute openfpga flow

omarghoneim3, 2022-11-14:
I created the xml file for architecture,VPR arch code, together with openfpga arch. Then, I edited spice netlist and verilog code. I get the error below which I don't know where it comes from

omarghoneim3, 2022-11-14:
python3 openfpga_flow/scripts/run_fpga_task.py basic_tests/full_testbench/configuration_chain --show_thread_logs
 INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task basic_tests/full_testbench/configuration_chain
 INFO (     MainThread) - Created "run116" directory for current task run
 INFO (     MainThread) - Running "yosys_vpr" flow
 INFO (     MainThread) - Found 2 Architectures 3 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 6 jobs
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) - Validating command line arguments
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) - Run directory : /home/omar/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run116/k4_N4_OpenFPGA_simulation/24clock.v/MIN_ROUTE_CHAN_WIDTH
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) - Running "yosys_vpr" Flow
 INFO (00_or2_MIN_ROUTE_CHAN_WIDTH) - Validating command line arguments
 INFO (00_or2_MIN_ROUTE_CHAN_WIDTH) - Run directory : /home/omar/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run116/k4_N4_OpenFPGA_simulation/or2/MIN_ROUTE_CHAN_WIDTH
 INFO (00_or2_MIN_ROUTE_CHAN_WIDTH) - Running "yosys_vpr" Flow
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) - Traceback (most recent call last):
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -   File "/home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 914, in <module>
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -     main()
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -   File "/home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 262, in main
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -     run_yosys_with_abc()
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -   File "/home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 604, in run_yosys_with_abc
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -     ys_params = create_yosys_params()
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -   File "/home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 480, in create_yosys_params
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -     tree = ET.parse(args.arch_file)
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -   File "/usr/lib/python3.6/xml/etree/ElementTree.py", line 1196, in parse
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -     tree.parse(source, parser)
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -   File "/usr/lib/python3.6/xml/etree/ElementTree.py", line 597, in parse
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) -     self._root = parser._parse_whole(source)
 INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) - xml.etree.ElementTree.ParseError: mismatched tag: line 283, column 12
ERROR (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_24clock.v_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "openfpga_flow/scripts/run_fpga_task.py", line 513, in run_single_script
subprocess.CalledProcessError: Command 'python3 /home/omar/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/omar/OpenFPGA/openfpga_flow/vpr_arch/k4_N4_OpenFPGA_simulation.xml /home/omar/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/24clock/24clock.v --top_module 24clock.v --run_dir /home/omar/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run116/k4_N4_OpenFPGA_simulation/24clock.v/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/omar/OpenFPGA/openfpga_flow/openfpga_shell_scripts/write_full_testbench_example_script.openfpga --openfpga_arch_file /home/omar/OpenFPGA/openfpga_flow/openfpga_arch/k4_N4_openfpga_arch.xml --openfpga_sim_setting_file /home/omar/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --openfpga_vpr_device_layout  --openfpga_fast_configuration  --power --power_tech /home/omar/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/omar/OpenFPGA/openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/run116/k4_N4_OpenFPGA_simulation/24clock.v/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --end_flow_with_test --TOP 24clock.v --CHAN_WIDTH 300 --READ_VERILOG_OPTIONS -nolatches' returned non-zero exit status 0.

tangxifan, 2022-11-17:
INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) - self._root = parser._parse_whole(source)
INFO (00_24clock.v_MIN_ROUTE_CHAN_WIDTH) - xml.etree.ElementTree.ParseError: mismatched tag: line 283, column 12

@omarghoneim3 Please check LINE 283 of your architecture files.

closed 2022-11-17 by tangxifan