lnis-uofu/OpenFPGA #912: Different multiplexers on LUTs and Interconnections

mustafaarslan0, 2022-11-22:
Hi. I was browsing the netlist and I realized multiplexers use much config memory than usual. Generally 2x1 MUX use 1 select bit like MUXes in the LUTs of the OpenFPGA. But other MUXes use as many selects as the number of inputs. What is the reason of that? Is it possible to build complete connections(crossbar) with MUX like LUT's MUX? Thanks.

The MUX using in crossbar:
[image]

The MUX using in LUT:
[image]

tangxifan, 2022-11-22:
@mustafaarslan0 

Routing multiplexers may be implemented in different circuit topology for FPGAs. This is because that each topology carries a different level of P.P.A.
Therefore, OpenFPGA support various circuit implementation of routing multiplexers, allowing chip designers to custom the circuit design of their routing multiplexers in different blocks of an FPGA. 
It means that it is chip designers' responsibility to determine which circuit designs they would like to pick. 
The architecture that you are using is a representative example of FPGA, where routing multiplexers are implemented in two-level structures. This is a common practice for modern commercial FPGAs. Here are some reference which may help you understand the reasons behind:

- https://si2.epfl.ch/~demichel/publications/archive/2017/07797202.pdf
- https://dl.acm.org/doi/abs/10.1007/s11265-007-0141-y

OpenFPGA's documentation about MUX structure support:

- https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_model_examples/#multiplexers

If you are working on a project with limited support on custom cells, I refer you to consider standard cell MUX2. See example: 

https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/mux_design/stdcell_mux2/config/task.conf

tangxifan, 2022-11-23:
@mustafaarslan0 Close for now. If you need more information, feel free to comment.

closed 2022-11-23 by tangxifan

mustafaarslan0, 2022-11-24:
Thanks for your reply. I have another question for muxes:
When "add_const_input" is true, does OpenFPGA selects const input if no input is selected? Or, is cont input exist for only parasitic activities and leakage power?

tangxifan, 2022-11-24:
The constant input is a physical input of routing mutliplexers when enabled. It is not virtual.