lnis-uofu/OpenFPGA #932: Does openfpga support the following crossbar？

narutozxp, 2022-12-05:
When I was looking for a local routing architecture with 50% connectivity, I found that there are many places in VTR that have descriptions as shown in the figure below. In my opinion，wouldn't this create a physical conflict.
For this way of writing, can OpenFPGA automatically derive a mux, or does it not support this way of writing?
[image]

narutozxp, 2022-12-05:
the <complete> tag and <direct> has a same output.

tangxifan, 2022-12-05:
@narutozxp It may cause errors in OpenFPGA. If you want to build a sparse crossbar, I suggest you to look into examples：

https://github.com/lnis-uofu/OpenFPGA/blob/1f0f13ce0d1be583aee2733c61eb4d308e45e6b0/openfpga_flow/vpr_arch/k4_N5_tileable_pattern_local_routing_40nm.xml#L284-L297

https://github.com/lnis-uofu/OpenFPGA/blob/1f0f13ce0d1be583aee2733c61eb4d308e45e6b0/openfpga_flow/vpr_arch/k6_frac_N10_tileable_adder_register_scan_chain_depop50_40nm.xml#L626-L646

tangxifan, 2023-01-09:
Close the issue as the crossbar is considered as illegal in physical representation.

closed 2023-01-09 by tangxifan