lnis-uofu/OpenFPGA #935: Can OpenFPGA support Flash based technologies such as Microchip FPGAs?

mbaykenar, 2022-12-05:
Is it possible FPGA fabric to have non-volatile configuration capability for OpenFPGA generated netlists?

tangxifan, 2022-12-05:
@mbaykenar I believe it is possible if you use non-volatile standard cell/custom cell to build the configuration memory element. OpenFPGA can integrate your cells to FPGA fabric. See details at 

https://openfpga.readthedocs.io/en/master/manual/arch_lang/config_protocol/#cmdoption-arg-circuit_model_name

https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_model_examples/#id8

mbaykenar, 2022-12-20:
Thank you very much. I assume most of the properties related to configuration and I/O are specific to the cell library that we aim to tape-out the FPGA IC such as programmable termination resistance, programmable drive strength and programmable I/O delay which are now standard properties for any commercial FPGA family.

tangxifan, 2022-12-20:
@mbaykenar You are right. However, I/Os in commercial FPGAs may be a complicated IP, which can operate in many different modes. Users can use bitstream to adjust the drive strength, direction, delay etc. of each I/O. OpenFPGA is designed to support all these features in programmable I/O blocks. My suggestion is to develop your own I/O block and integrate it just other programmable hard IPs, e.g., DSP and BRAM. OpenFPGA just need to know the configuration bits to program the I/O in each proper mode. Users are responsible for the quality of their I/O design.

tangxifan, 2023-01-09:
Close for now. If there is any follow-up question, feel free to post.

closed 2023-01-09 by tangxifan