lnis-uofu/OpenFPGA #955: How to load bitstream to FPGA generated by OpenFPGA

mbaykenar, 2022-12-20:
> **Describe the bug**

> **Additional context**
> Add any other context about the problem here.

I looked for documentation in both OpenFPGA and SOFA but could not find how to load bitstream to the FPGA IC. I can find bitstream format is plain text (.bit) and for scan_chain configuration it is a configuration information for 1 line. After the generation of .bit file, how can I load this file to FPGA?

In SOFA, there is CCFF_HEAD and PROG_CLK, which I assume CCFF_HEAD is pushed to configuration chain shift-reg at each rising edge of PROG_CLK. So what kind of HW & SW is available to load the bitstream? 

Does a simple 2-wire clk and data pin capable HW and related SW running in a microcontroller or a logic design in an FPGA to sent data with clock is enough?

Regards,

tangxifan, 2022-12-20:
@mbaykenar To download bitstream, you may consider to develop your firmware by referring to the [full testbench](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#write-full-testbench) generated by OpenFPGA. To ease the development and debugging for your firmware, you may consider to use another FPGA to program your own FPGA.

tangxifan, 2023-02-09:
Close this issue for now. If you have follow-up questions, feel free to reopen and post.

closed 2023-02-09 by tangxifan