lnis-uofu/OpenFPGA #960: change FPGA architecture of "basic_tests/fixed_pins" example

shkim1104, 2022-12-28:
I tried to change the openfpga architecture file to "k4_frac_N8_tileable_register_scan_chain_nonLR_caravel_io_skywater130nm"
and edit fix_pins.place file according to my application ports
the first picture is task.conf, the second picture is led_fix_pins.place and the third picture is application

[image]

[image]

[image]

when run the task, error occur not to generate fabric and bitstream.

error message:
[image]

How could I configure pins to the "**k4_frac_N8_tileable_register_scan_chain_nonLR_caravel_io_skywater130nm**" architecture?

tangxifan, 2022-12-29:
@shkim1104 Can you share the blif file and openfpgashell.log under the runtime directory, for example, 
``openfpga_flow/tasks/basic_tests/fix_pins/run004/k4_frac_N8_tileable_register_scan_chain_nonLR_caravel_io_skywater130nm/led1/MIN_ROUTE_CHAN_WIDTH``

shkim1104, 2022-12-29:
Dear tangxifan,

I attached ~/MIN_ROUTE_CHAN_WIDTH directory.
Thank you for analyzing the problem.

Thank you,
Sohyeon Kim

-- 

=====================================================
*Sohyeon Kim*, Ph.D. Student

*D*igital *S*ystem *A*rchitecture *L*ab. (link
<https://urldefense.proofpoint.com/v2/url?u=http-3A__dsa.ewha.ac.kr_&d=DwMGaQ&c=vY6N1du11amUSmqXT2U6Yqw6I-si6nWulSErsUHFzRg&r=1LQGmfBDKYyCp6ld6Uxl-Q&m=nGMR6SlMjut800qf5pxNFvRaWNR9L-cx36rrfdEZuHA&s=6wPPtOUrHnDarA2IJxviercNf73xEC46vA3Yov6Txt4&e=>
)

Dept. of Electronic and Electrical Engineering

Ewha Womans University

Mobile : +82-10-2958-6937
E-mail : ***@***.***

=====================================================

tangxifan, 2022-12-29:
@shkim1104 Can you double check the attachments? I cannot find any.

shkim1104, 2022-12-29:
I attached .zip format file again.
If you still don't see the file, please let me know.

Thank you,
Sohyeon Kim

-- 

=====================================================
*Sohyeon Kim*, Ph.D. Student

*D*igital *S*ystem *A*rchitecture *L*ab. (link
<https://urldefense.proofpoint.com/v2/url?u=http-3A__dsa.ewha.ac.kr_&d=DwMGaQ&c=vY6N1du11amUSmqXT2U6Yqw6I-si6nWulSErsUHFzRg&r=1LQGmfBDKYyCp6ld6Uxl-Q&m=nGMR6SlMjut800qf5pxNFvRaWNR9L-cx36rrfdEZuHA&s=6wPPtOUrHnDarA2IJxviercNf73xEC46vA3Yov6Txt4&e=>
)

Dept. of Electronic and Electrical Engineering

Ewha Womans University

Mobile : +82-10-2958-6937
E-mail : ***@***.***

=====================================================

tangxifan, 2022-12-29:
@shkim1104 Still cannot see the attachments. Can you try to upload by using Github webpage? Email may not work well.

shkim1104, 2022-12-30:
I uploaded files in the following link.
https://github.com/shkim1104/openfpga_fixpins

Thank you,
Sohyeon Kim

-- 

=====================================================
*Sohyeon Kim*, Ph.D. Student

*D*igital *S*ystem *A*rchitecture *L*ab. (link
<https://urldefense.proofpoint.com/v2/url?u=http-3A__dsa.ewha.ac.kr_&d=DwMGaQ&c=vY6N1du11amUSmqXT2U6Yqw6I-si6nWulSErsUHFzRg&r=1LQGmfBDKYyCp6ld6Uxl-Q&m=nGMR6SlMjut800qf5pxNFvRaWNR9L-cx36rrfdEZuHA&s=6wPPtOUrHnDarA2IJxviercNf73xEC46vA3Yov6Txt4&e=>
)

Dept. of Electronic and Electrical Engineering

Ewha Womans University

Mobile : +82-10-2958-6937
E-mail : ***@***.***

=====================================================

tangxifan, 2022-12-30:
@shkim1104 It is due to the routing failures. 

```
Cannot route from clb.O[14] (RR node: 110 type: SOURCE location: (1,1) class: 32 capacity: 1) to io_right[0].outpad[0] (RR node: 290 type: SINK location: (3,1) class: 12 capacity: 1) -- no possible path
Failed to route connection from 'LED[0]' to 'out:LED[0]' for net 'LED[0]' (#2)
Routing failed.
···

Please give a try with a bigger routing channel width by changing the task configuration file

```
openfpga_vpr_route_chan_width=40
```

shkim1104, 2022-12-30:
I did as you said and it worked!
Thank you :)

Thank you,
Sohyeon Kim

-- 

=====================================================
*Sohyeon Kim*, Ph.D. Student

*D*igital *S*ystem *A*rchitecture *L*ab. (link
<https://urldefense.proofpoint.com/v2/url?u=http-3A__dsa.ewha.ac.kr_&d=DwMGaQ&c=vY6N1du11amUSmqXT2U6Yqw6I-si6nWulSErsUHFzRg&r=1LQGmfBDKYyCp6ld6Uxl-Q&m=nGMR6SlMjut800qf5pxNFvRaWNR9L-cx36rrfdEZuHA&s=6wPPtOUrHnDarA2IJxviercNf73xEC46vA3Yov6Txt4&e=>
)

Dept. of Electronic and Electrical Engineering

Ewha Womans University

Mobile : +82-10-2958-6937
E-mail : ***@***.***

=====================================================

tangxifan, 2022-12-31:
Close for now. If there are any follow-up questions, please feel free to raise.

closed 2022-12-31 by tangxifan