lnis-uofu/OpenFPGA #985: Merge fix present in VPR

alaindargelas, 2023-01-09:
> **Describe the bug**
> VPR contains a fix on verbosity level of Packer message:

https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/b1bab3d31862f8cf030ab1c202dab7fa6ddab446

@tangxifan, We desperately need that fix to be merged in OpenFPGA as it generates 100Mb log files that could be avoided with the fix.
Please also add @serge-rs as OpenFPGA developer so we can track/inquire about fixes. 

[X ] VPR 

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: <The problem commit id>
> 2. Execute OpenFPGA task or your own example: <more details>
> 3. See error

> **Expected behavior**
> A clear and concise description of what you expected to happen.

> **Screenshots**
> If applicable, add screenshots to help explain your problem.

> **Enviornment (please complete the following information):**

>  - OS: 

>  - Compiler:

> - Version:

> **Additional context**
> Add any other context about the problem here.

tangxifan, 2023-01-09:
@alaindargelas Just updated VTR feature branch. Will need another 1 day to propagated to OpenFPGA.

tangxifan, 2023-01-09:
@alaindargelas @serge-rs Invitation has been sent. Please double check.

alaindargelas, 2023-01-09:
Thanks @tangxifan

closed 2023-01-11 by alaindargelas