lnis-uofu/OpenFPGA #1002: How to change write_fabric_verilog SRC/ path?

alain-rs, 2023-01-17:
In **https://github.com/os-fpga/FOEDAG/blob/d7785ae03e822253105538a4d887fb9a4bdc9409/src/Compiler/CompilerOpenFPGA.cpp#L2303**

We instruct OpenFPGA to write the netlists necessary for bitstream verification. 

In the fabric_netlists.v produced there are paths like:
/ ------ Include user-defined netlists -----
`include "./SRC/CustomModules/clock_inverter.v"

Is there a way to direct OpenFPGA to use a different prefix than SRC/?

ganeshgore, 2023-01-17:
Yes `--file` option does that
https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#write-fabric-verilog

alain-rs, 2023-01-17:
Check the code line I point to above, the --file is used to point to BIT_SIM/ to output the files.
My question is for the other files pointer to by SRC/, that seems to be hardcoded and yet needs to be customized. 

// ------ Include defines: preproc flags -----
`include "BIT_SIM/fpga_defines.v"

// ------ Include user-defined netlists -----
`include "./SRC/CustomModules/clock_inverter.v"
`include "./SRC/CustomModules/inverter.v"
`include "./SRC/CustomModules/logic0.v"
`include "./SRC/CustomModules/logic1.v"
`include "./SRC/sc_verilog/tcbn16ffcllbwp7d5t16p96cpd.v"
`include "./SRC/CustomModules/rs_ioff.v"
`include "./SRC/CustomModules/rs_ff.v"
`include "./SRC/CustomModules/rs_xor_mux2.v"
`include "./SRC/CustomModules/rs_ccff.v"
`include "./SRC/CustomModules/rs_preio.v"
`include "./SRC/CustomModules/rs_dsp.v"
`include "./SRC/CustomModules/rs_bram.v"
// ------ Include primitive module netlists -----
`include "BIT_SIM/sub_module/inv_buf_passgate.v"
`include "BIT_SIM/sub_module/arch_encoder.v"
`include "BIT_SIM/sub_module/local_encoder.v"

ganeshgore, 2023-01-17:
Those other paths are coming from the Openfpga architecture XML file, please look for `verilog_netlist` parameter in the `circuit_model` tags

alaindargelas, 2023-01-17:
I see, thanks.

alaindargelas, 2023-01-17:
You can close the issue

closed 2023-01-17 by ganeshgore