lnis-uofu/OpenFPGA #1025: OpenFPGA generated efpga prototyping and chip implementation issues

shkim1104, 2023-01-30:
Thanks for making such a great tool, "OpenFPGA". It is very helpful for my research.

I'm trying to implement OpenFPGA-generated efpga as a chip, and I'm writing this to ask how to solve the problems I encountered in the process.

1. I want to protopyte the efpga fabric generated with the openFPGA tool on xilinx FPGA through vivado. However, it needs about 22,000 luts of Xilinx FPGA resources to upload an 8x8 homogeneous structure eFPGA that has 560 lut resources, and synthesis time is too long.
And eventually Implementation and bitstream generation fails due to combinational loop.
Is there any effective way to prototyping efpga created with OpenFPGA on xilinx FPGA with vivado?

2. When I create an efpga with a homogeneous structure and perform a fuction simulation with vcs or xrun, it takes a long time (6 hours) to configure the bitstream of a 36x36 size efpga and even verify the application.
Is there any way to save time?

3. When eFPGA generated by openfpga is synthesized in 180nmCMOS process and gate-level simulation is run, if bitstream which is generated about sequential logic application (FIFO) is configured, it runs only at 10MHz frequency.
I wonder if it is possible to increase the maximum frequency further, if so, what is the maximum frequency for each process (65nm CMOS, 28nm CMOS , etc.).
Also, I wonder how to apply .sdc that can give a constraint to the operational frequency when generating a bitstream, and an example template.
As stated in the openfpga documentation, even if I put it in the form of synopsys synthetic constraint in the application sdc path, the openfpga task does not recognize the app.sdc file. how could I solve the issue?

thank you,

tangxifan, 2023-01-30:
@shkim1104 Thank you for your interests in OpenFPGA.

- We do not have any suggestions to prototyping the FPGA fabrics on an Xilinx FPGA. I did see some guys tried a 2x2 using a Xilinx FPGA, and it is working fine. Bigger fabric may not work. In general the combination loops are the key barriers
- If you are experience long runtime in design verification, can you confirm if you are using a preconfigured testbench or a full testbench (see details in [doc](https://openfpga.readthedocs.io/en/master/manual/fpga_verilog/testbench/)). I suggest you to try preconfigured testbenches first. It should take around 5-15 minutes for the fabric size you have, depending on your HDL simulators. Do not use iVerilog for the FPGA in your size.
- To optimize the performance, you need to use PrimeTime to extract timing for your FPGA layout. Then annotate it to architecture file. Afterwards, you can get a realistic timing estimation for your benchmark. Do not trust any timing in existing architecture on OpenFPGA, they are not for your FPGA fabric. SDC is a complicated task. We are still investigate it.

tangxifan, 2023-02-09:
Close this issue for now. If you have follow-up questions, feel free to reopen and post.

closed 2023-02-09 by tangxifan