lnis-uofu/OpenFPGA #1032: [WIP] Support programmable and dedicated networks for global signals

tangxifan, 2023-02-02:
**Is your feature request related to a problem? Please describe.**
Global signals, such as ``clock``, ``reset`` and ``set``, span a complete or a considerable portion of FPGA fabrics.
From a physical design perspective, the global signals require a dedicated network to route a source to all the sinks.
Common practice to implement the networks includes:

- H-tree
- Fishbone

Each implementation carries different consideration on latency and skews, which are highly dependent on the product requirements.

Currently, in the fabric netlists that are auto-generated by OpenFPGA do not include any network for global signals.
It assumes that physical designers should implement the networks through tools, such as Synopsys ICC2 or Cadence Innovus.
However, in recent researches, we see that *Clock Tree Synthesis* in modern ASIC tools cannot efficiently implement these networks in a reasonable runtime [1].
Researchers also show that adding pre-routed networks to fabric netlists is a practical solution, with which layout of a 100k-LUT FPGA can be accomplished in 24 hours.

Given the importance of the networks, it is necessary to support the dedicated networks in OpenFPGA's fabric generator.
It avoids manual and massive netlist changes before physical design, which 

- is designed just for a specific FPGA architecture. When changing architectures, existing scripts may require considerable changes
- causes additional work for design verification
- carries a risk of verification failures which may take long time to debug

[1] Ganesh Gore, Xifan Tang, and Pierre-Emmanuel Gaillardon. 2021. A Scalable and Robust Hierarchical Floorplanning to Enable 24-hour Prototyping for 100k-LUT FPGAs. In Proceedings of the 2021 International Symposium on Physical Design (ISPD '21). Association for Computing Machinery, New York, NY, USA, 135–142. https://doi.org/10.1145/3439706.3447047

**Describe the solution you'd like**

tangxifan, 2023-02-02:
### Changes on user interface

Users can select if a programmable network is needed for each global signal, by adding XML description to the OpenFPGA architecture. 
New XML syntax will be introduced to the [Physical Tile Annotation](https://openfpga.readthedocs.io/en/master/manual/arch_lang/annotate_vpr_arch/#physical-tile-annotation)

```
<tile_annotations>
  <global_port name="<string>" is_clock="<bool>" is_reset="<bool>" is_set="<bool>" default_val="<int>">
    <network type="none|htree|fishbone" entry_x="<int>" entry_y="<int>"/>
    <tile name="<string>" port="<string>" x="<int>" y="<int>"/>
    ...
  </global_port>
</tile_annotations>
```

ganeshgore, 2023-02-02:
Hello @tangxifan, thank you for this proposal. 
Can you elaborate on how this syntax will help create different clock tree structures in different layouts? 
Let's first take an example of 4x4 homogenous fabric then 10x8, and then some small heterogenous version.

tangxifan, 2023-02-02:
@ganeshgore No problem. 4x4 is a regular size, we do not expect any issue. You can refer to the figures in the first post.

ganeshgore, 2023-02-03:
Issue is not 4x4, its other layouts I listed 10x8, and some heterogenous design

tangxifan, 2023-02-03:
> Issue is not 4x4, its other layouts I listed 10x8, and some heterogenous design

ok. Let me elaborate this.