lnis-uofu/OpenFPGA #1044: SDF file

zliu1Charlotte, 2023-02-07:
I am curious if OpenFPGA can generate SDF files?

zliu1Charlotte, 2023-02-07:
BTW is the generated FPGA fabric netlist a gate-level netlist? 
Thanks.

tangxifan, 2023-02-09:
> I am curious if OpenFPGA can generate SDF files? or is it possible to know the timing from LUT output to a FF input inside of a CLB? I am curious to know the post-p&r's timing from LUT to FF, and FF to LUT.

You may refer to the VPR post-synthesis support. You call vpr in openfpga shell. See details in 
 https://docs.verilogtorouting.org/en/latest/tutorials/timing_simulation/

tangxifan, 2023-02-09:
> BTW is the generated FPGA fabric netlist a gate-level netlist? Thanks.

OpenFPGA offers many options for the netlist generation. You can generate RTL or gate-level netlists, even a mix of them. You can select the Verilog style for each circuit model. 

See syntax ``dump_structural_verilog`` in https://docs.verilogtorouting.org/en/latest/tutorials/timing_simulation/

closed 2023-02-09 by tangxifan

zliu1Charlotte, 2023-02-09:
Thanks! Can we use skywater tech on k4_N4_tileable_40nm.xml when using  `vpr k4_N4_tileable_40nm.xml xor_cipher.blif --gen_post_synthesis_netlist on`

tangxifan, 2023-02-09:
> Thanks! Can we use skywater tech on k4_N4_tileable_40nm.xml when using `vpr k4_N4_tileable_40nm.xml xor_cipher.blif --gen_post_synthesis_netlist on`

Yes. You should be able to.

tangxifan, 2023-02-09:
> when we do --gen_post_synthesis_netlist, can we set the clk period? what clk period is the sdf based on? Thanks

Please check VPR's documentation about this. Give a try on SDC file.

zliu1Charlotte, 2023-02-09:
I want to 

> 

Do you have an example on this? So far in my openfpga shell script, I do

vpr ${VPR_ARCH_FILE} ${VPR_TESTBENCH_BLIF} --clock_modeling route --gen_post_synthesis_netlist on --timing_report_detail netlist

I believe my  ${VPR_ARCH_FILE} is arch0=${PATH:OPENFPGA_PATH}/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml

How do i tweak it to skywater? 

I checked one example k4_frac_N8_tileable_register_scan_chain_nonLR_caravel_io_skywater130nm.xml
But I do not see major differences between k4_frac_N8_tileable_register_scan_chain_nonLR_caravel_io_skywater130nm.xml and k4_N4_tileable_40nm.xml.

Thanks

zliu1Charlotte, 2023-02-09:
Thank you. 
I checked the link https://docs.verilogtorouting.org/en/latest/tutorials/timing_simulation/#timing-simulation-tutorial 
I understand that sdf is generated by providing arch.xml and .blif. 
When you say try it on the SDC file, how should I place the SDC files in my command, it should be as an input?
You are probably pointing to this `global_ports.sdc`
```
set_units -time ps

##################################################
# Create clock                                    
##################################################
create_clock -name clk[0] -period 20000 -waveform {0 10000} [get_ports {clk[0]}]
##################################################
# Create programmable clock                       
##################################################
create_clock -name prog_clk[0] -period 20000 -waveform {0 10000} [get_ports {prog_clk[0]}]
```