lnis-uofu/OpenFPGA #1056: gen_post_synthesis_netlist on skywater130nm

zliu1Charlotte, 2023-02-13:
Hi,

I tried using vpr ${VPR_ARCH_FILE} ${VPR_TESTBENCH_BLIF} --clock_modeling route --gen_post_synthesis_netlist on 

And it worked.

My VPR_ARCH_FILE is k4_N4_tileable_40nm.xml

But I assume if we want to generate the post synthesis netlist (the .v and the .sdf) using skywater 130nm, how should i work this out, or is this even possible?

zliu1Charlotte, 2023-02-13:
I took a look inside of some skywater.xml examples in vpr_arch. 
My guess is that for the sdf generation, the delay_constant seems like a key factor.
And it seems different among 40nm 90nm and 130nm..

tangxifan, 2023-05-02:
@zliu1Charlotte The timing numbers come from your timing extraction results. To get accurate timing values, you need to perform physical design, timing extraction and back-annotation. The timing values in these arch file may be very out-of-dated.