lnis-uofu/OpenFPGA #1095: How to avoid Constant-0 drivers added to undriven nets

cristianomodesti, 2023-03-13:
### Discussed in https://github.com/lnis-uofu/OpenFPGA/discussions/1089

<div type='discussions-op-text'>

<sup>Originally posted by **cristianomodesti** March  7, 2023</sup>
Good morning to everyone.
I am trying to synthesise a very simple HDL design composed by a LSFR which basically implements a shift register whose input bit is a linear function of the previous content of the register. I made this test just to debug some strange behaviour I got during the implementation of a bigger design.

A hint of this test design can be found in the following code snippet:
``` systemverilog
module lsfr(
    input  logic clk_i,
    input  logic res_n_i,
    input  logic run_i,
    output logic data_o0,
    output logic data_o1,
    output logic data_o2,
    output logic data_o3,
    output logic data_o4,
    output logic data_o5,
    output logic data_o6,
    output logic data_o7,
    output logic data_o8,
    output logic data_o9,
    output logic data_o10,
    output logic data_o11,
    output logic data_o12,
    output logic data_o13,
    output logic data_o14,
    output logic data_o15,
    output logic data_o16,
    output logic data_o17,
    output logic data_o18,
    output logic data_o19
);

  localparam p_size = 10;//20;
  localparam [p_size-1:0] p_init = 0;

  logic [p_size-1:0] r_lfsr;

  logic c_lfsr_msb;

  always_comb c_lfsr_msb = r_lfsr[p_size-10] ~^ r_lfsr[p_size-07];
        
  always_ff @(posedge clk_i, negedge res_n_i) begin
    if (!res_n_i)begin
      r_lfsr <= p_init;
    end else begin
      if (run_i) begin
        r_lfsr <= {c_lfsr_msb, r_lfsr[p_size-1:1]};
      end
    end
  end

  assign data_o0  = |r_lfsr;
  assign data_o1  = r_lfsr[0];
  assign data_o2  = r_lfsr[1];
  assign data_o3  = r_lfsr[2];
  assign data_o4  = r_lfsr[0];
  assign data_o5  = r_lfsr[0];
  assign data_o6  = r_lfsr[0];
  assign data_o7  = r_lfsr[0];
  assign data_o8  = r_lfsr[0];
  assign data_o9  = r_lfsr[0];
  assign data_o10 = r_lfsr[3];
  assign data_o11 = r_lfsr[4];
  assign data_o12 = r_lfsr[5];
  assign data_o13 = r_lfsr[0];
  assign data_o14 = r_lfsr[0];
  assign data_o15 = r_lfsr[0];
  assign data_o16 = r_lfsr[0];
  assign data_o17 = r_lfsr[0];
  assign data_o18 = r_lfsr[0];
  assign data_o19 = r_lfsr[0];
endmodule
```

RTL simulation works fine and I get the expected output signals but when I simulate the generated FPGA netlist, I am getting the expected behaviour just for `data_o0 - data_o3` and `data_o10 - data_o12`, whereas the other outputs are seen as non-driven and thus yosys adds a constant-0 driver.

The point is that I need those "copy" output signals having the same values (`r_lsfr[0]`) without having them treated like undriven nets. 
Has anybody ever had an issue like this? How can I tell yosys these are needed signals?

Thanks for the help!</div>

tangxifan, 2023-05-02:
@cristianomodesti I suggest to transfer the issue to Yosys github repo, where you can get full support.

cristianomodesti, 2023-05-02:
> @cristianomodesti I suggest to transfer the issue to Yosys github repo, where you can get full support.

I did exactly that and they game me the very same answer. To transfer it to OpenFPGA. But thanks anyway