lnis-uofu/OpenFPGA #1158: OpenFPGA does not support customizing Yosys synthesis parameters

narutozxp, 2023-05-09:
I defined a parameter called “bench_yosys_adder_map_verilog_common” in the task.conf file and used“YOSYS_ADDER_MAP_VERILOG” in the Yosys script template file. However, the parameter was not recognized. 
Is there any way that I can use to enable customizing Yosys synthesis parameters?
Thanks!

tangxifan, 2023-05-12:
@narutozxp Please double check your yosys template script (.ys), see if you have such variable defined.

https://github.com/lnis-uofu/OpenFPGA/blob/2ce26c1c71a1ce57beeafb4334ce4f494cad3b39/openfpga_flow/tasks/basic_tests/k4_series/k4n4_fracff/config/task.conf#L35

narutozxp, 2023-05-12:
the contents of our task.conf  are as follows:
```bash
bench_yosys_cell_sim_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_fracff_40nm_cell_sim.v
bench_yosys_dff_map_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_fracff_40nm_dff_map.v
bench_yosys_bram_map_rules_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_40nm_bram.txt
bench_yosys_bram_map_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_40nm_bram_map.v
bench_yosys_dsp_map_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_40nm_dsp_map.v
bench_yosys_dsp_map_parameters_common=-D DSP_A_MAXWIDTH=36 -D DSP_B_MAXWIDTH=36 -D DSP_A_MINWIDTH=2 -D DSP_B_MINWIDTH=2 -D DSP_NAME=mult_36x36
bench_read_verilog_options_common = -nolatches

# customizing yosys script
bench_yosys_adder_map_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/techmap_adder.v
bench_yosys_adder_sim_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/efpga_adder_sim.v
bench_yosys_frac_float_sim_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/frac_float_cells.v
#specify the yosys script templete
bench_yosys_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/memory_adder_opt_multi_mode_float.ys
```
and the contents of our yosys template script are as follows:
```bash
# Read verilog files
read_verilog ${READ_VERILOG_OPTIONS} ${VERILOG_FILES}
# Read technology library
read_verilog -lib -specify ${YOSYS_CELL_SIM_VERILOG}
read_verilog -lib -specify ${YOSYS_FRAC_FLOAT_SIM_VERILOG}
read_verilog -lib -specify ${YOSYS_ADDER_SIM_VERILOG}
```
the `${YOSYS_CELL_SIM_VERILOG}` could be replaced by our definition, but the others didn't be recongnized

narutozxp, 2023-05-12:
@tangxifan the contents of our task.conf  are as follows:
```bash
bench_yosys_cell_sim_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_fracff_40nm_cell_sim.v
bench_yosys_dff_map_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_fracff_40nm_dff_map.v
bench_yosys_bram_map_rules_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_40nm_bram.txt
bench_yosys_bram_map_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_40nm_bram_map.v
bench_yosys_dsp_map_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/k6_frac_N10_tileable_adder_chain_dpram8K_dsp36_40nm_dsp_map.v
bench_yosys_dsp_map_parameters_common=-D DSP_A_MAXWIDTH=36 -D DSP_B_MAXWIDTH=36 -D DSP_A_MINWIDTH=2 -D DSP_B_MINWIDTH=2 -D DSP_NAME=mult_36x36
bench_read_verilog_options_common = -nolatches

# customizing yosys script
bench_yosys_adder_map_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/techmap_adder.v
bench_yosys_adder_sim_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/efpga_adder_sim.v
bench_yosys_frac_float_sim_verilog_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/frac_float_cells.v
#specify the yosys script templete
bench_yosys_common=${PATH:OPENFPGA_PATH}/narutozxp/work/study/config_dir/k4n8_tactic_dsp/opt_adder_memory_frac_float/config/memory_adder_opt_multi_mode_float.ys
```
and the contents of our yosys template script are as follows:
```bash
# Read verilog files
read_verilog ${READ_VERILOG_OPTIONS} ${VERILOG_FILES}
# Read technology library
read_verilog -lib -specify ${YOSYS_CELL_SIM_VERILOG}
read_verilog -lib -specify ${YOSYS_FRAC_FLOAT_SIM_VERILOG}
read_verilog -lib -specify ${YOSYS_ADDER_SIM_VERILOG}
```
the `${YOSYS_CELL_SIM_VERILOG}` could be replaced by our definition, but the others didn't be recongnized

tangxifan, 2023-05-12:
@narutozxp I see. Please try to add your variable name to the following line:

https://github.com/lnis-uofu/OpenFPGA/blob/2ce26c1c71a1ce57beeafb4334ce4f494cad3b39/openfpga_flow/scripts/run_fpga_task.py#L294-L312

narutozxp, 2023-05-16:
Thanks, it really helps a lot

tangxifan, 2023-06-21:
The problem has been resolved. Now you can use ${YOSYS_ADDER_MAP_VERILOG} to specify the adder mapping file.

closed 2023-06-21 by tangxifan