lnis-uofu/OpenFPGA #1173: Configuration memory design with ql_memory_bank configuration protocol

mustafaarslan0, 2023-05-22:
I am working on ql_memory_bank configuration protocol and I have a few questions about it:

1. I know that analog circuits (read/write circuitry, sense amplifiers) are required for read/write operation on SRAM bit line. How should be these circuits is handled on physical design?
2. All SRAM cells are spread in the submodules. Is there any way to gather SRAM cells on tile-level? I think gathered SRAM cells can be replaced with SRAM memory macro easily.

tangxifan, 2023-05-22:
@mustafaarslan0 

1. The analog circuits may be considered during physical design. Currently, OpenFPGA does include decoders, shift registers as peripheral for the QL memory banks.
2. We are planning on this feature. I think it is a good idea. But it will take a lot of effort.

mustafaarslan0, 2023-08-29:
Thank you for answer and development.

closed 2023-08-29 by mustafaarslan0