lnis-uofu/OpenFPGA #1197: Fail with benchmarks using

anhdv2000, 2023-06-04:


anhdv2000, 2023-06-04:
I have a benchmark
```
module dff01 (
  input clk,    // Clock
  input reset_n,
  input  in,
  output reg  out
  
);
  always @(posedge clk or negedge reset_n) begin 
    if(~reset_n) begin
      out <= 1;
    end else begin
      out <= in;
    end
  end
endmodule : dff01
```
i run with test **tasks/basic_tests/k4_series/k4n4_fracff**. But vpp fail output Q_fpga isn't assgined 1 when reset active

tangxifan, 2023-06-20:
My suggestion is to create specific FF models where you want active-low set.

closed 2023-06-20 by tangxifan

anhdv2000, 2023-06-20:
@tangxifan If I understand correctly, for each benchmark we run, we need to define the appropriate flip-flop in the openfpga.xml file using the following line: <pb_type name="clb.fle[physical].fabric.ff" circuit_model_name="MULTI_MODE_DFFRQ" mode_bits="0"/>.
If that's the case, for benchmarks that contain multiple types of flip-flops, we would also need a configurable flip-flop model that can operate like all the different types of flip-flops, depending on the configuration.