lnis-uofu/OpenFPGA #1199: Config cell parameter in "k6_frac_N10_tileable_adder_chain_40nm" in OpenFPGA

Nguyenthehuy1, 2023-06-05:
Hello everyone,

I want to find out how to config cell parameter in "k6_frac_N10_tileable_adder_chain_40nm" in OpenFPGA by entering at OpenFPGA root:      "vi openfpga_flow/openfpga_arch/k6_frac_N10_adder_chain_40nm_openfpga.xml". But I cannot see any documents about cell parameters in this. 
 
Where can I find it? 

Thank you, everyone.
Best regard
 Huy

tangxifan, 2023-06-13:
Hi @Nguyenthehuy1 ,
  Can you elaborate what the cell parameters are in these files?

Nguyenthehuy1, 2023-06-14:
I want to change "Technology library" in this architecture xml file. I want
to change my own vdd, pn_ratio, min_witdh, chan_length.... but i wonder
what path need I insert?( default: OpenFGPA/ openfpga_flow/tech/PTM_45nm/
45nm.pm? ), and the delay_matrix parametter? i have 7x7 matrix, which point
it mapping?( nxn ??)

Thank you

tangxifan, 2023-06-15:
I believe you can find the answers in #1211

Nguyenthehuy1, 2023-06-15:
Thank you so much for responding to me, 

I have replaced some cells with “sky_water_130.v” (illustrated below) in the “K6_frac_N10_adder_chain_40nm_openfpga.xml” file. And I have two questions: “ 

1. How can I map “delay_matrix” parameter in
  “OpenFPGA/yosys-plugins/clockgating plugin/tests/regfile/lib/sky130_fd_sc_hd.lib”. 

2. As you said in #1211, is it means we should remain the initial value in “Technology_library (including pn_ratio, min_width ...)? ( so what do those value means?)”, and how we mapping parameter with “sky_water..” library? 

*********************************************************
*** code xml is below:
`<?xml version="1.0"?>

<openfpga_architecture>
  <technology_library>
    <device_library>
      <device_model name="logic" type="transistor">
        <lib type="industry" corner="TOP_TT" ref="M" path="${OPENFPGA_PATH}/openfpga_flow/tech/PTM_45nm/45nm.pm"/>
        <design vdd="0.9" pn_ratio="2"/>
        <pmos name="pch" chan_length="40e-9" min_width="140e-9" variation="logic_transistor_var"/>
        <nmos name="nch" chan_length="40e-9" min_width="140e-9" variation="logic_transistor_var"/>
      </device_model>
      <device_model name="io" type="transistor">
        <lib type="academia" ref="M" path="${OPENFPGA_PATH}/openfpga_flow/tech/PTM_45nm/45nm.pm"/>
        <design vdd="2.5" pn_ratio="3"/>
        <pmos name="pch_25" chan_length="270e-9" min_width="320e-9" variation="io_transistor_var"/>
        <nmos name="nch_25" chan_length="270e-9" min_width="320e-9" variation="io_transistor_var"/>
      </device_model>
    </device_library>
    <variation_library>
      <variation name="logic_transistor_var" abs_deviation="0.1" num_sigma="3"/>
      <variation name="io_transistor_var" abs_deviation="0.1" num_sigma="3"/>
    </variation_library>
  </technology_library>
  <circuit_library>
    <circuit_model type="inv_buf" name="INVTX1" prefix="INVTX1" is_default="true">
      <design_technology type="cmos" topology="inverter" size="1"/>
      <device_technology device_model_name="logic"/>
      <port type="input" prefix="in" size="1"/>
      <port type="output" prefix="out" size="1"/>
      <delay_matrix type="rise" in_port="in" out_port="out">
        10e-12
      </delay_matrix>
      <delay_matrix type="fall" in_port="in" out_port="out">
        10e-12
      </delay_matrix>
    </circuit_model>

      <circuit_model type="inv_buf" name="sky130_fd_sc_ls__inv_1" prefix="sky130_fd_sc_ls__inv_1" verilog_netlist="/home/dolphin/huynt0/SoC/OpenFPGA/skywater-pdk/libraries/sky130_fd_sc_ls/latest/cells/inv/sky130_fd_sc_ls__inv_1.v" dump_structural_verilog="true">
       <design_technology type="cmos" topology="inverter" size="3"/>
       <device_technology device_model_name="logic"/>
       <port type="input" prefix="A" size="1"/>
       <port type="output" prefix="Y" size="1"/>
     </circuit_model>

    <circuit_model type="inv_buf" name="sky130_fd_sc_ls__bufinv_8" prefix="sky130_fd_sc_ls__bufinv_8" verilog_netlist="/home/dolphin/huynt0/SoC/OpenFPGA/skywater-pdk/libraries/sky130_fd_sc_ls/latest/cells/bufinv/sky130_fd_sc_ls__bufinv_8.v">
      <design_technology type="cmos" topology="buffer" size="1" num_level="2" f_per_stage="4"/>
      <device_technology device_model_name="logic"/>
      <port type="input" prefix="A" size="1"/>
      <port type="output" prefix="Y" size="1"/>
      <delay_matrix type="rise" in_port="A" out_port="Y">
        10e-12
      </delay_matrix>
      <delay_matrix type="fall" in_port="A" out_port="Y">
        10e-12
      </delay_matrix>

    </circuit_model>
    <circuit_model type="inv_buf" name="tap_buf4" prefix="tap_buf4" is_default="false">
      <design_technology type="cmos" topology="buffer" size="1" num_level="3" f_per_stage="4"/>
[... 170 lines cut ...]
    <pb_type name="clb.fle[n2_lut5].ble5.lut5" physical_pb_type_name="clb.fle[physical].fabric.frac_logic.frac_lut6" mode_bits="01" physical_pb_type_index_factor="0.5">
      
      <port name="in" physical_mode_port="in[0:4]"/>
      <port name="out" physical_mode_port="lut5_out[0:0]" physical_mode_pin_rotate_offset="1"/>
    </pb_type>
    <pb_type name="clb.fle[n2_lut5].ble5.ff" physical_pb_type_name="clb.fle[physical].fabric.ff"/>
    
    <pb_type name="clb.fle[arithmetic].arithmetic.lut4" physical_pb_type_name="clb.fle[physical].fabric.frac_logic.frac_lut6" mode_bits="11" physical_pb_type_index_factor="0.25">
      
      <port name="in" physical_mode_port="in[0:3]"/>
      <port name="out" physical_mode_port="lut4_out[0:0]" physical_mode_pin_rotate_offset="1"/>
    </pb_type>
    <pb_type name="clb.fle[arithmetic].arithmetic.adder" physical_pb_type_name="clb.fle[physical].fabric.adder"/>
    <pb_type name="clb.fle[arithmetic].arithmetic.ff" physical_pb_type_name="clb.fle[physical].fabric.ff"/>
    
    <pb_type name="clb.fle[n1_lut6].ble6.lut6" physical_pb_type_name="clb.fle[physical].fabric.frac_logic.frac_lut6" mode_bits="00">
      
      <port name="in" physical_mode_port="in[0:5]"/>
      <port name="out" physical_mode_port="lut6_out"/>
    </pb_type>
    <pb_type name="clb.fle[n1_lut6].ble6.ff" physical_pb_type_name="clb.fle[physical].fabric.ff" physical_pb_type_index_factor="2" physical_pb_type_index_offset="0"/>
    
  </pb_type_annotations>
</openfpga_architecture>
`

tangxifan, 2023-06-20:
See my answer in #1221

closed 2023-06-20 by tangxifan