lnis-uofu/OpenFPGA #1211: How to change parameter of cell when replace library

Nguyenthehuy1, 2023-06-08:
Hello everyone,

I want to replace the original library with my library. Still, I can not find anything about the timing, power, or ... and where the original library inserts the file library (with .lib)?

Thank you
Best regards
Huy

tangxifan, 2023-06-13:
Hi @Nguyenthehuy1 
   We do not need a .lib file when running OpenFPGA. In the openfpga architecture file, you may define the name of circuit models  using the name of standard cell or any other cell from your library. Then the output netlists are mapped to your technology library.

Consider the example: https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_model_examples/#standard-cell-multiplexer
If you have a MUX standard cell, e.g., MUX2D1, you can name the circuit model as

```
<circuit_model type="mux" name="MUX2D1" prefix="MUX2D1">
```

Nguyenthehuy1, 2023-06-14:
Thank you so much for responding to me,

1. As you said, I do need a *.lib file, so do we need a *.db file? If not, how openFPGA architecture can understand timings or delays..., because “verilog_neslist” is just about function, 

2. and if I change “vdd, pn_ratio.. in technology_library in K6_fac_N10_adder_chanin_40nm_openfaga.xml”, where I can find this change (output)? 

Thank you

tangxifan, 2023-06-15:
You do not need a .db file. We do not handle delay or timing through .db file. We expect designers to back-annotate timing to our architecture file. See the delay parameters in the VPR architecture description: 

https://docs.verilogtorouting.org/en/latest/arch/reference/#tag-%3Cdelay_constantmax=

The VDD, pn ratio etc. are required for FPGA-SPICE. If you are not using it, just leave dummy values.

Nguyenthehuy1, 2023-06-15:
Thank you for responding to me,

what about delay_matrix values? What do you rely on to determine those values?
In a .lib file, we usually have rise and fall time to determine its pointing to n x n matrix, for example: (if index_1 = 0.01 (position 1 of index 1), and index_2 = 0.0005 (position 1 of index 2) => we have the value: 1x1 (position of value))

index_1("0.0100000000, 0.0230505800, 0.0531329300, 0.1224745000, 0.2823108000, 0.6507428000, 1.5000000000");
index_2("0.0005000000, 0.0012265480, 0.0030088420, 0.0073809800, 0.0181062600, 0.0444164000, 0.1089577000");
values("0.0335820000, 0.0324435000, 0.0295460000, 0.0224904000, 0.0052049000, -0.0370580000, -0.1406482000", \
           "0.0330832000, 0.0319430000, 0.0290717000, 0.0220002000, 0.0046753000, -0.0375541000, -0.1412124000", \
                        "0.0325117000, 0.0313173000, 0.0285060000, 0.0214064000, 0.0040939000, -0.0381634000, -0.1417224000", \
                        "0.0319710000, 0.0307878000, 0.0279193000, 0.0208373000, 0.0035437000, -0.0386644000, -0.1422429000", \
                        "0.0314140000, 0.0302145000, 0.0274018000, 0.0205042000, 0.0033479000, -0.0388361000, -0.1425158000", \
                        "0.0309295000, 0.0297733000, 0.0269733000, 0.0200110000, 0.0026365000, -0.0393728000, -0.1428567000", \
                        "0.0334235000, 0.0323313000, 0.0294552000, 0.0222344000, 0.0049136000, -0.0379047000, -0.1425150000");

closed 2023-06-20 by tangxifan