lnis-uofu/OpenFPGA #1219: OpenFPGA seems doesn't support binding VTR's short type switch

narutozxp, 2023-06-15:
> **Describe the bug**

As my architecture file had the following content, especially the short type switch named adder_chain_wire, i needed to bind the switch to physical circuit. However, to my knowledge, OpenFPGA just supports mux type switch and direct connection in the directlist tag. Is there any helpful information? 
Thanks!
```xml
<switchlist>
  <!-- mux_trans_size Specifies the size (in minimum width transistors) of each transistor in the two-level mux used by mux type switches. -->
  <!-- buf_size If set to auto, sized automatically from the R value. This allows you to use timing models without R’s and C’s and still be able to measure area. -->
  <switch type="mux" name="sb_mux" R="551" Cin="0.77e-15" Cout="4e-15" mux_trans_size="2.630740" buf_size="27.645901">
    <Tdel num_inputs="12" delay="${SB_MUX_DELAY}"/>
    <Tdel num_inputs="15" delay="${SB_MUX_DELAY}"/>
    <Tdel num_inputs="20" delay="${SB_MUX_DELAY}"/>
  </switch>

  <!-- Not currently allowing an ipin cblock switch to have fanin dependent values 
    refer: home/data/shawn/eFPGA/Code/OpenFPGA/OpenFPGA/vtr-verilog-to-routing/vpr/src/base/CheckArch.cpp Line: 66 -->
  <!-- <switch type="mux" name="cb_in_mux" R="2231.5" Cout="0" Cin="1.47e-15" mux_trans_size="1.222260" buf_size="auto">
    <Tdel num_inputs="12" delay="${CB_MUX_DELAY}"/>
    <Tdel num_inputs="15" delay="${CB_MUX_DELAY}"/>
    <Tdel num_inputs="20" delay="${CB_MUX_DELAY}"/>
  </switch> -->
  <switch type="mux" name="cb_in_mux" R="2231.5" Cout="0" Cin="1.47e-15" Tdel="${CB_MUX_DELAY}" mux_trans_size="1.222260" buf_size="auto"/>

  <switch type="short" name="adder_chain_wire" R="0" Cout="0" Cin="0" Tdel="${CARRY_CHAIN_DELAY}"/>
</switchlist>
```

tangxifan, 2023-06-17:
Correct. We will catch this feature soon.

tangxifan, 2023-06-17:
You do not need to worry about the switch binding. You can specify the circuit model for the short switch using OpenFPGA's architecture description language:

https://github.com/lnis-uofu/OpenFPGA/blob/cd748b1a5e75e9d32422ced63ad5a46137c414a1/openfpga_flow/openfpga_arch/k6_frac_N10_adder_register_chain_40nm_openfpga.xml#L214

narutozxp, 2023-06-25:
@tangxifan  Yep, i indeed did that as what you said , but the following description may incur problems as mentioned in https://github.com/lnis-uofu/OpenFPGA/issues/1174#issuecomment-1562196049, i.e. resulting the inconsistent between the fabric generated by OpenFPGA and the routing resource graph generated by origin VPR.
```xml
<direct name="adder_carry" circuit_model_name="direct_interc" type="column" x_dir="positive" y_dir="positive"/>