lnis-uofu/OpenFPGA #1223: Benchmark with special flip-flops

anhdv2000, 2023-06-20:
Can openfpga run benchmarks with special flip-flops as shown below
```
always @(posedge clk or negedge resetb) begin 
	if(~resetb) begin
		Q <= 1; // Q assigned to 1
	end else begin
		Q <= D;
	end
end
```
or 
```
always @(posedge clk or negedge resetb) begin 
	if(~resetb) begin
		Q <= A; // Q assigned to variable
	end else begin
		Q <= Q - 1;
	end
end
```
I notice that you haven't provided any examples of benchmarks like this. Is it because you lack examples or is it because openfpga doesn't support this?

tangxifan, 2023-06-20:
@anhdv2000 Please check this example: https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/tasks/basic_tests/k4_series/k4n4_fracff/config

OpenFPGA supports many types of flip-flops. You can find fruitful examples. If you want, you can create your own FF mapping rules for yosys.

anhdv2000, 2023-06-20:
@tangxifan yes. I checked it. And I noticed that your examples only involve changing the active edge (posedge reset, negedge reset ...) but do not mention the different output behavior of flip-flops when reset is active,  as in the two examples I mentioned above.
I think we need to have a flip-flop model that can support all types of flip-flops to define in the openfpga.xml file.
Have you ever run benchmarks like the two examples I mentioned above? And is it possible to update such examples on the repository?

tangxifan, 2023-06-20:
@anhdv2000 Correct. As an FPGA architect, users should be aware of the flip-flop models in their FPGA architecture. 
- You need to define all the flip-flop models that you want to support. So that yosys and vpr can map them properly.
- We have provided a few examples in the [example architecture](https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/vpr_arch/k4_frac_N4_tileable_fracff2edge_40nm.xml): 
 
  - FF triggered at positive clock edge
  - FF triggered at negative clock edge
  - FF with an active high reset and triggered at positive clock edge
  - FF with an active high reset and triggered at negative clock edge
  - FF with an active low reset and triggered at positive clock edge
  - FF with an active low reset and triggered at negative clock edge

Based on what I see, what you need is a FF model with an active low set and triggered at positive clock edge. I believe you can easily develop one by referring to existing examples.
We would like to provide more examples to the community. However, our current focus is more on #1220 etc. We will try to provide more examples after reaching these milestones.

anhdv2000, 2023-06-28:
Hi @tangxifan , I have a new issue with running a preconfigured testbench with the benchmark as DFF, like the code below
```
always @(posedge clk)
begin
    if (en) 
        Q0 <= D0;
end
```
FPGA ports displays 1'bx.
[image]
I believe the reason for this issue is that the output Q0 is not initialized
However, when I run it with the full testbench, it works correctly
Is there any way for me to run it with the preconfigured testbench instead of the full testbench? Running the full testbench takes a lot of time