lnis-uofu/OpenFPGA #1244: OpenFPGA failed with basic benchmark

anhdv2000, 2023-07-05:
Hi @tangxifan , I have an issue with a fairly basic benchmark for you
I have a benchmark
```
module test (
  input clk,    
  input reset_n, 
  input a,
  input b,
  output c,
  output reg c_reg
);
 assign c = a & b;
 always @(posedge clk or negedge reset_n) begin 
   if(~reset_n) begin
     c_reg <= 0;
   end else begin
     c_reg <= c;
   end
 end
endmodule : test
```
I run with test **tasks/basic_tests/k4_series/k4n4_fracff** but failed (output of FPGA is x).  I am unable to determine whether the reason for the issue lies with the VPR tool or if there was an error in generating the bitstream. Can you run this test and let me know the reason for the failure?

tangxifan, 2023-07-26:
@anhdv2000  Can you check your .blif file which is outputted by yosys? If you see any failures in verification, the best way is to use GTKwave to check the waveform and see if the clock and reset signals are correctly driving the FFs inside the FPGA.

tangxifan, 2023-07-26:
@anhdv2000 Since you do need a active-low reset, you may need to use pin constraints when generating Verilog testbenches. See an example here:

https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/basic_tests/k4_series/k4n4_fracff/config/pin_constraints_resetb.xml