lnis-uofu/OpenFPGA #1248: Bug in bintoi_charvec() function

chungshien, 2023-07-11:
> In ql_memory_bank with flatten BL/WL configuration protocol, we will need to

1. set the BL address - https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga/src/fpga_bitstream/fabric_bitstream.cpp#L192

    BL is basically a string (more accurately vector of char), which the string could contains 'x', '0', '1'. In this particular use case, normally  BL contains a series don't care bit 'x' and only one '1' bit. Example: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx1xxxxxxxxxxxxxxxxxxxxxxxxxxx

2. When we set the BL, we will convert the string into a vector of uint64_t. https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga/src/fpga_bitstream/fabric_bitstream.cpp#L170

    The function will break a long string of BL into multiple of 64 chars (or less), each 64 chars will be converted into two uint64_t: for represent data, one represent dont-care-mask. 

3. To convert the chars chunk into uint64_t, we will eventually call https://github.com/lnis-uofu/OpenFPGA/blob/master/libs/libopenfpgautil/src/openfpga_decode.cpp#L172-L182

    Per my testing, this function does not perform the conversion correctly. For example if we have 64 chars of '1', "1111111111111111111111111111111111111111111111111111111111111111", the expectation of result would be 0xFFFFFFFFFFFFFFFF, but I found that the result is 0. 

   Refer below simple example. 

> **To Reproduce**
1. See this simple example code

[image]

2. ret0 is code that is similar to OpenFPGA implementation. ret1 is my workaround where I cast the pow() result to size_t first before addition. 

3. See the printf result. The result start to becoming problem at i=53, and eventually 64 characters of '1' will become 0 instead of expected result (0xFFFFFFFFFFFFFFFF)

[image]

> **Enviornment (please complete the following information):**

>  - OS: 

>  - Compiler:

> - Version:

alaindargelas, 2023-07-14:
@chungshien, please submit a PR with the fix. @tangxifan will review.

chungshien, 2023-07-16:
Created a PR - https://github.com/lnis-uofu/OpenFPGA/pull/1252

chungshien, 2023-08-11:
PR is in. Close issue.

closed 2023-08-11 by chungshien