lnis-uofu/OpenFPGA #1249: Incorrect Switch-Box generation in tileable architectures

Crossbomber, 2023-07-13:
> **Expected behavior**

According to "A study on switch block patterns for tileable fpga routing architectures", the switch boxes in tileable Architectures should connect in the same directions as non-tileable architectures. Passing wire segments should be connected to Multiplexers in orthogonal directions, ending wire segments should be connected to Multiplexers on all 3 Sides. When tileable is turned off, this is how the generated switch boxes look in my tests.

> **Describe the bug**

The switch-boxes generated when using a tileable architecture do not look as expected. 
For passing wire segments, the signal is propagated straight through and to a Multiplexer in orthogonal directions (as expected), but also in a multiplexer in the straight through direction. 
For ending wire segments, the signal is only led to multiplexers in orthogonal directions, but not in a multiplexer straight through.

This way multiplexers are larger when using long wire segments and poorly routable when using short wire segments. When using segments of length 1, straight connections spanning multiple CLBs are impossible, resulting in a slow "zigzag" routing.

This can be observed with all architectures, if tileable is set to true. I have not found another parameter which affects this. 

The build is from late january, but i have not noticed any major changes in the tileable switch box generation of the source code which could affect this since then.

If my results stem from incorrect usage of the tool, i would be thankful for help in finding the cause of this issue. If instead these results are what is to be expected, please provide information on why this contradicts the information in "A study on switch block patterns for tileable fpga routing architectures". Thank you.

tangxifan, 2023-07-26:
@Crossbomber This looks a bug if what you see happens to any central switch blocks. Switch blocks on perimeter can be strange as you see since they only have 2 sides or 3 sides of pins. Can you provide an example? I can take a look.

Thank you.

Crossbomber, 2023-07-28:
Hello, 

thank you for your response. 

The described problem also appears on the central switch blocks. As an example, i ran the basic_tests/generate_fabric task. To demonstrate it, i made the following changes:
- fixed the layout to a width and height of 5, so central switch blocks are generated
- added the write_gsb_to_xml command to the shell script
- changed the length of the segment "L4" to 1, and adjusted the switch and connection block population accordingly (sb pattern 1 1 and cb pattern 1)

The switch block at x=1, y=1 (not on the perimeter) looks like this:
```xml
<rr_sb x="1" y="1" num_sides="4">
	<CHANY side="top" index="0" node_id="1451" segment_id="0" segment_name="L4" mux_size="3" sb_module_pin_name="chany_top_out">
		<driver_node type="OPIN" side="top" index="0" node_id="468" grid_side="right" sb_module_pin_name="top_bottom_grid_right_width_0_height_0_subtile_0__pin_O_1_"/>
		<driver_node type="CHANX" side="right" index="3" node_id="1214" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_right_in"/>
		<driver_node type="CHANX" side="left" index="0" node_id="1191" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_left_in"/>
	</CHANY>
	<CHANY side="top" index="2" node_id="1453" segment_id="0" segment_name="L4" mux_size="3" sb_module_pin_name="chany_top_out">
		<driver_node type="OPIN" side="top" index="1" node_id="469" grid_side="right" sb_module_pin_name="top_bottom_grid_right_width_0_height_0_subtile_0__pin_O_5_"/>
		<driver_node type="CHANX" side="right" index="5" node_id="1216" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_right_in"/>
		<driver_node type="CHANX" side="left" index="18" node_id="1209" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_left_in"/>
	</CHANY>
.
.
.
```

Notice how only channels from the left and right side are connected with the top output channels.

For comparison, here is part of the same switch block when setting tileable to "false":
```xml
<rr_sb x="1" y="1" num_sides="4">
	<CHANY side="top" index="0" node_id="1451" segment_id="0" segment_name="L4" mux_size="4" sb_module_pin_name="chany_top_out">
		<driver_node type="OPIN" side="top" index="0" node_id="274" grid_side="right" sb_module_pin_name="top_bottom_grid_right_width_0_height_0_subtile_0__pin_O_1_"/>
		<driver_node type="CHANX" side="right" index="3" node_id="1214" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_right_in"/>
		<driver_node type="CHANY" side="bottom" index="0" node_id="1431" segment_id="0" segment_name="L4" sb_module_pin_name="chany_bottom_in"/>
		<driver_node type="CHANX" side="left" index="0" node_id="1191" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_left_in"/>
	</CHANY>
	<CHANY side="top" index="2" node_id="1453" segment_id="0" segment_name="L4" mux_size="4" sb_module_pin_name="chany_top_out">
		<driver_node type="OPIN" side="top" index="1" node_id="278" grid_side="right" sb_module_pin_name="top_bottom_grid_right_width_0_height_0_subtile_0__pin_O_5_"/>
		<driver_node type="CHANX" side="right" index="5" node_id="1216" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_right_in"/>
		<driver_node type="CHANY" side="bottom" index="2" node_id="1433" segment_id="0" segment_name="L4" sb_module_pin_name="chany_bottom_in"/>
		<driver_node type="CHANX" side="left" index="18" node_id="1209" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_left_in"/>
	</CHANY>
.
.
.
```

Here, channels from the left, right and bottom are connected with the output at the top, allowing for straight through connections. With longer segments, the problem is less noticable, but persists. Here is the same switch block with the original length 4 wire segments and with a tileable architecture:
```xml
<rr_sb x="1" y="1" num_sides="4">
	<CHANY side="top" index="0" node_id="1311" segment_id="0" segment_name="L4" mux_size="15" sb_module_pin_name="chany_top_out">
		<driver_node type="OPIN" side="top" index="0" node_id="468" grid_side="right" sb_module_pin_name="top_bottom_grid_right_width_0_height_0_subtile_0__pin_O_1_"/>
		<driver_node type="OPIN" side="top" index="3" node_id="471" grid_side="right" sb_module_pin_name="top_bottom_grid_right_width_0_height_0_subtile_0__pin_O_13_"/>
		<driver_node type="OPIN" side="top" index="6" node_id="562" grid_side="left" sb_module_pin_name="top_bottom_grid_left_width_0_height_0_subtile_0__pin_O_7_"/>
		<driver_node type="OPIN" side="top" index="9" node_id="565" grid_side="left" sb_module_pin_name="top_bottom_grid_left_width_0_height_0_subtile_0__pin_O_19_"/>
		<driver_node type="CHANX" side="right" index="1" node_id="1166" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_right_in"/>
		<driver_node type="CHANX" side="right" index="9" node_id="1174" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_right_in"/>
		<driver_node type="CHANX" side="right" index="15" node_id="1186" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_right_in"/>
		<driver_node type="CHANX" side="right" index="17" node_id="1182" segment_id="0" segment_name="L4" sb_module_pin_name="chanx_right_in"/>
		<driver_node type="CHANY" side="bottom" index="0" node_id="1291" segment_id="0" segment_name="L4" sb_module_pin_name="chany_bottom_in"/>
		<driver_node type="CHANY" side="bottom" index="8" node_id="1299" segment_id="0" segment_name="L4" sb_module_pin_name="chany_bottom_in"/>
[... cut ...]
I have noticed this with all architectures, when tileable is set to true. I have rebuilt openfpga on a clean machine yesterday running Ubuntu 22.04, but the behaviour hasnt changed.

I hope this information helps you. A zip file with the gsb-xml files for segment lengths 1, 2 and 4 both for tileable and non tileable architectures can be found here: [SB_1_1_GSB_XML.zip](https://github.com/lnis-uofu/OpenFPGA/files/12196776/SB_1_1_GSB_XML.zip)

I can provide further details if you need them. Thank you for your help.

closed 2023-11-14 by tangxifan

tangxifan, 2023-11-14:
@Crossbomber This issue has been resolved as #1439 is merged. Now you can enable option ``concat_wire`` in your needs.

See details in 

https://openfpga.readthedocs.io/en/master/manual/arch_lang/addon_vpr_syntax/#cmdoption-arg-concat_wire