lnis-uofu/OpenFPGA #1269: Problem with flipflop architecture

Nguyenthehuy1, 2023-07-28:
I have written my own flipflop architecture (it includes posedge clk -  posedge reset - posedge set - active  HIGH and   posedge clk -  posedge reset - posedge set - active  LOW), I define two modules in "openfpga_dff_sim.v" 
-> in the "ff_test_2.blif" file, I saw two architectures with the same name: “_DFF_PPP_”.

I wonder how open FPGA can understand that two flip-flop architecture.

tangxifan, 2023-07-28:
@Nguyenthehuy1 If you see the DFF_PPP in your blif file. It means that Yosys failed to do technology mapping for some DFFs in your RTL design. I suggest to double check your dff mapping rules. For example: 

https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/openfpga_yosys_techlib/openfpga_dff_map.v

Nguyenthehuy1, 2023-08-21:
Assume that DFF_PPP is true, how can I distinguish two architectures with the same name (DFF_PPP)?

tangxifan, 2023-09-06:
@Nguyenthehuy1 I believe Yosys documentation provides detailed explanation. The ``DFF_PPP`` is an internal DFF type of Yosys. If tech map failed, Yosys just write out the internal FF. Then you see errors in VPR.