lnis-uofu/OpenFPGA #1304: When I am running a TASK, I use VPR -- sdc_ file fir.sdc .But I got some errors

Chris202305, 2023-08-18:
> **Describe the bug**
I am a student who started to learn openfpga weeks ago and I don't how to solve the problem. Anyboby can help me?

ERROR 1
Type: SDC file
line: -1
message: syntax error,unexpected [ near '['

I use a Docker of openfpga.
> A clear and concise description of what the bug is.

fir.sdc
create_clock -name clk -period 10 -waveform {0 0.5} [get_ports clk]

vpr ${VPR_ARCH_FILE} ${VPR_TESTBENCH_BLIF} --clock_modeling route ${OPENFPGA_VPR_DEVICE_LAYOUT} --sdc_file fir.sdc

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: <The problem commit id>
> 2. Execute OpenFPGA task or your own example: <more details>
> 3. See error

> **Expected behavior**
> A clear and concise description of what you expected to happen.

> **Screenshots**
> If applicable, add screenshots to help explain your problem.

> **Enviornment (please complete the following information):**

>  - OS: 

ubuntu 20.04.5

>  - Compiler:

> - Version:

> **Additional context**
> Add any other context about the problem here.

tangxifan, 2023-08-19:
@Chris202305 Please refer to [VTR's SDC documentation](https://docs.verilogtorouting.org/en/latest/vpr/sdc_commands/)

Regarding your case, it looks like you want to define a virtual clock. And the ``[get_ports clk]`` may not work in VTR's SDC parser. Please double check.

Chris202305, 2023-08-21:
@tangxifan Thank you for your guidance. I take it for granted that I have used the format I used to write SDC here, which is really inappropriate. But sometimes there are some errors that I don't know where to search for relevant files, so I may consult you for further questions. Thank you for your patient guidance as a beginner.

Chris202305, 2023-08-21:
I encountered a new problem while running a task, why is the output_verilog.v generated by my task having an extra '\' which is unexpected before the port name? That means the port in my original design which is called data_ In, but when it comes to output_ verilog. v, it becomes  '\data_ In'.  This caused that the auto_check_tb  got some errors. Could you give me some guidance on where the possible problem lies?

Chris202305, 2023-08-21:
@tangxifan

Chris202305, 2023-08-22:
Thank you for your guidance. I take it for granted that I have used the format I used to write SDC here, which is really inappropriate. But sometimes there are some errors that I don't know where to search for relevant files, so I may consult you for further questions. Thank you for your patient guidance as a beginner.

I encountered a new problem while running a task, why is the output_verilog.v generated by my task having an extra '' which is unexpected before the port name? That means the port in my original design which is called data_ In, but when it comes to output_ verilog. v, it becomes '\data_ In'. This caused that the auto_check_tb got some errors. Could you give me some guidance on where the possible problem lies?

 

雨霖林
***@***.***

tangxifan, 2023-09-06:
@Chris202305 Yosys flattens the bus port during synthesis, as required by OpenFPGA flow. So you see the difference in port naming. There are two ways to bypass this：

- Use the implicit port mapping when generating the Verilog testbench (remove the option ``--explicit_port_mapping``). **This is a quick workaround**. You can find a working example here: 

https://github.com/lnis-uofu/OpenFPGA/blob/163e94c05c14881453674de999722d969ff94b4f/openfpga_flow/tasks/fpga_verilog/dsp/multi_mode_mult_16x16/config/task.conf#L19

https://github.com/lnis-uofu/OpenFPGA/blob/163e94c05c14881453674de999722d969ff94b4f/openfpga_flow/openfpga_shell_scripts/fix_heterogeneous_device_pbPinFixup_example_script.openfpga#L58C8-L60

**or**

- Use the bus group file. This is a better and nice solution but requires some effort to understand.

https://github.com/lnis-uofu/OpenFPGA/blob/163e94c05c14881453674de999722d969ff94b4f/openfpga_flow/tasks/basic_tests/bus_group/auto_gen_bus_group/config/task.conf#L19-L28