lnis-uofu/OpenFPGA #1311: Clock Network Description

narutozxp, 2023-08-22:
The clock network in OpenFPGA(descripted in [OpenFPGA docs](https://openfpga.readthedocs.io/en/master/manual/file_formats/clock_network/#general-settings)) is different with VTR(descripted in [VTR docs](https://docs.verilogtorouting.org/en/latest/arch/reference/#clocks)), so is there any examples to show the usage of OpenFPGA's clock networks.
Thanks!

closed 2023-09-06 by tangxifan

narutozxp, 2023-09-19:
@tangxifan Thanks for your kind answer. However, according to the example, the clock network description is not in the vpr architecture file, but in a single file. Then, OpenFPGA read the network description file by the command of "read_openfpga_clock_arch -f filename". As a result, if i have a clock network description used to generate netlist by OpenFPGA, the VPR can't use such a file to routing clk signal as it is not descripted in vpr architecture file.
 In my opinion,to solve this problem, i should have the same description in vpr architecture file,. However, as mentioned above, OpenFPGA and VPR have different grammar. So, it puzzles me a lot, and is there any suggestions about this?
Thanks!

tangxifan, 2023-09-27:
@narutozxp OpenFPGA's clock arch description is developed to model a programmable clock network from a physical design perspective. It models how the clock network is exactly implemented in an FPGA fabric. From a user's point of view, you do not need to know such complexity. That is why you see VPR's clock modeling is different. The way is to go is that you craft your clock network and then crafte a simpilified modeling in VPR.

narutozxp, 2023-10-26:
@tangxifan, However, there is still a problem, if i craft a programmable clock network and then create a simplified modeling in VPR. it means that even though  VPR could generate the correct routing results, OpenFPGA may not produce a valid bitstream.
Because it involves the routing and config bits between clock network and general routing.