lnis-uofu/OpenFPGA #1318: ql_memory_bank shift_register-decoder mixed protocol

mustafaarslan0, 2023-08-25:
**Is your feature request related to a problem? Please describe.**
Currently, ql_memory_bank protocol has 3 different protocol: Flatten, Decoder, Shift register. 

- Flatten is the fastest one but it needs too much I/O. I think it is well working protocol.
- I inspected simulations that Decoder is improper for sram/latch configuration cells because it programs one bit on BL per WL and enables same WL for another BL bit. This causes that previous programmed bit in the same BL goes unknown. I think WL's must be enabled once and all bits in the same BL must be programmed per WL.
- Shift register is good too but it is the slowest one in this trio.

**Describe the solution you'd like**
I think this trio should be mixed in one protocol like this:

- Shift register on BL: Bit line is not suitable for decoder type so it can be shift_register to reduce I/O.
- Decoder on WL: Word line is suitable for decoder type so it can be decoder to reduce programming time.
- Shift register and decoder outputs should be same with Flatten protocol.

I tried this suggestion with custom modifications on final netlist and it works well. I think it is considerable. Can it be done? And can it be multi region?

Regards.

tangxifan, 2023-09-06:
@mustafaarslan0 Make sense to me. The mixed protocol should be done. I will see if it can be made by the end of this year. Otherwise, we can target next year. If you need it urgently, we can discuss.

Here are my thoughts about these protocols:

- The flatten is native way. It is designed for developers to debug and quickly spot any problems on BL/WLs. As you said, it requires a lot of I/Os. In practice, chip designers build their own protocols on top of the flatten BL/WLs, outside the FPGA fabric.  The protocol gives chip designers 100% freedom to explore their best configuration protocol (external interface) for their FPGA chip.
- Decoder is indeed a problem. What we want is to configure all the BLs in one row when a WL is enabled. Therefore, it is not efficient for BLs but still a solution for WLs.
- Shift registers actually have been used in some commercial tape-outs already. If you see 'X' in bitstream, they are don't care bits. You can switch them to [0|1] through an option [keep_dont_care_bits](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_bitstream_commands/#cmdoption-keep_dont_care_bits). You can even customize the shift register chains through fabric key file. See [details](https://openfpga.readthedocs.io/en/master/manual/file_formats/fabric_key/#bl-shift-register-banks). It minimizes the I/O numbers for BL/WLs and easy to handle timing. Also, it saves engineers a lot of time and effort to build their own protocol and testbenches.

mustafaarslan0, 2023-09-07:
@tangxifan nice to see your thoughts. It is not urgent but it would be nice alternative to shift registers. If my architecture does not work with shift registers, it will be only solution for me and I have to do some custom works. <br>I saw keep_dont_care_bits is not enabled in my OpenFPGA script but there are X's on bitstream. I will investigate shift registers again over keep_dont_care_bits and fabric key. I will write results. <br>Regards.

mustafaarslan0, 2023-10-12:
@tangxifan I tried shift_register protocol and it works fine but there are X's in bitstream despite disabled keep_dont_care_bits option (according the guide: [keep_dont_care_bits](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_bitstream_commands/#cmdoption-keep_dont_care_bits)). However it works fine either if there are X's or convert X's to 0's. 

Also I modified the configuration section of auto generated testbench because it works buggy. I would like to develop auto generated testbench verilogs but I dont know how to do it. Can you guide for that?

Due to slow configuration time and difficulties of shift registers, I appreciate that if you develop the mixed protocol which I mentioned in https://github.com/lnis-uofu/OpenFPGA/issues/1318#issue-1866644160.

Regards.

tangxifan, 2023-10-14:
@mustafaarslan0 If you are interested in the source codes of testbench generator, you may refer to

- https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga/src/fpga_verilog/verilog_top_testbench.cpp
- https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga/src/fpga_verilog/verilog_formal_random_top_testbench.cpp

Regarding the slow configuraton time, I suppose that you are referrig to the full testbench? 

We are currently very tight on resources. Our priorities are mainly on the following topics:

- #1371 
- #1032 
- #905 
- #1242 

If you would like to prioritize the development, it is better that we establish business-level collaboration.

mustafaarslan0, 2023-10-17:
@tangxifan I have not advanced software background but I will inspect source codes. 

Regarding the slow configuration time, yes I mention about the full testbench because it is closer the ASIC implementation.

No need the change your priorities. I have respect for your priorities and developments. Thank you.

closed 2023-10-17 by mustafaarslan0