lnis-uofu/OpenFPGA #1341: Unconnected Clock

AhmadHouraniah, 2023-09-06:
> **Describe the bug**
> The generated fabric creates a secondary clock signal that is undriven and unconnected.

 Which part of OpenFPGA is buggy 

FPGA-Verilog 

> **To Reproduce**
> Steps to reproduce the behavior:
Run the generate_fabric and generate_testbench tasks with --clock_modeling ideal in the openfpga shell script

> **Expected behavior**
I expect to see only two clocks (a programming clock and a global clock).
> **Screenshots**
[image]

> **Enviornment (please complete the following information):**

>  - OS: Ubuntu 18.04 
>  - Compiler: gcc-8

> - Version: 1.2.0, also tested with current master, same behaviour was observed.

> **Additional context**
> This issue doesn't seem to affect the output, the fabric still behaves as expected and it passes the full testbench. However, it is still strange to have this undriven and unconnected clock signal. This clock can be seen in the clb blocks and its submodules, finally reaching logical_tile_clb_mode_default__fle_mode_physical__fabric_mode_default__ff.v 
This module then instantiates the flop using the global clock instead and leaves the ff_clk signal unconnected.

tangxifan, 2023-09-06:
@AhmadHouraniah 

This is due to the ``global_port`` is defined for FF circuit models in your openfpga architecture. For example:

https://github.com/lnis-uofu/OpenFPGA/blob/163e94c05c14881453674de999722d969ff94b4f/openfpga_flow/openfpga_arch/k6_frac_N8_stdcell_mux_40nm_openfpga.xml#L145

When enabled, OpenFPGA will create dedicated wires to connect the clock port (or any global port defined), from the primitive to the top-level. In such case, the clock port that is defined by VPR architecture will not be used. 

If you do want to see a clean netlist, you may try to use the global tile port feature. See the example: 

https://github.com/lnis-uofu/OpenFPGA/tree/master/openfpga_flow/tasks/basic_tests/global_tile_ports/global_tile_clock/config

AhmadHouraniah, 2023-09-07:
Following the example you provided, I generated a clean netlist that doesn't contain the undriven clock.
Thank you very much.

closed 2023-09-07 by AhmadHouraniah