lnis-uofu/OpenFPGA #1375: Implicit Net declaration

nachiket, 2023-09-25:
I'm trying to use dc_shell to compile the Verilog code generated by OpenFPGA. I'm getting the following errors:

On my local task I do the following:
```
python3 $OPENFPGA_ROOT/openfpga_flow/scripts/run_fpga_task.py synthesizable_verilog
setenv RTLDIR synthesizable_verilog/latest/k6_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH
dc_shell-xg-t -f cbox.synth.tcl
```

Lots of implicit net declaration errors
```
Compiling source file ./synthesizable_verilog/latest/k6_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH/SRC/sub_module/muxes.v
Error:  ./synthesizable_verilog/latest/k6_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH/SRC/sub_module/muxes.v:15: Implicit net declaration for 'in' prohibited by '`default_nettype none' compiler directive. (VER-943)
Error:  ./synthesizable_verilog/latest/k6_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH/SRC/sub_module/muxes.v:15: Implicit net declaration for 'sram' prohibited by '`default_nettype none' compiler directive. (VER-943)
Error:  ./synthesizable_verilog/latest/k6_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH/SRC/sub_module/muxes.v:15: Implicit net declaration for 'sram_inv' prohibited by '`default_nettype none' compiler directive. (VER-943)
Error:  ./synthesizable_verilog/latest/k6_N10_tileable_40nm/and2/MIN_ROUTE_CHAN_WIDTH/SRC/sub_module/muxes.v:15: Implicit net declaration for 'out' prohibited by '`default_nettype none' compiler directive. (VER-943)
*** Presto compilation terminated with 4 errors. ***
```

tangxifan, 2023-09-25:
@nachiket Now we allow users to specify if the default net type through a command line option ``--default_nettype``

https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#cmdoption-default_net_type

Please note that if you select ``none``, the netlist size will be much bigger than when you select ``wire``

nachiket, 2023-09-25:
Thanks for the quick reply. 

I’m currently using the `run_fpga_task.py` command. Where do I provide this flag?
Can I supply this in the task.conf file?

tangxifan, 2023-09-25:
@nachiket To enable the option, you need to craft your openfpga shell script (if you are currrently using a default openfpga shell script in OpenFPGA). You can refer to the following example:

```
https://github.com/lnis-uofu/OpenFPGA/blob/e345b97b848d0b425044c66fd76173693fca05e3/openfpga_flow/openfpga_shell_scripts/verilog_default_net_type_example_script.openfpga#L47
```

And in your task configuration file, you can define the parameter:

https://github.com/lnis-uofu/OpenFPGA/blob/e345b97b848d0b425044c66fd76173693fca05e3/openfpga_flow/tasks/fpga_verilog/verilog_netlist_formats/explicit_port_mapping_default_nettype_wire/config/task.conf#L19-L22

As such, you can freely switch between different options in your task.

tangxifan, 2023-09-25:
Can you check which task you are calling? Then you can spot the task.conf file. The shell script is defined in the ``openfpga_shell_template`` line of your task.conf file

nachiket, 2023-09-25:
Oh, I also recall having to manually patch SDC files, I assume that’s not needed anymore?

This is what I used to do but didn’t need to do it now. Not sure if the timing analysis just work without these fixes.

Minor naming adjustment to leaf-level (`SDC_leaf`), and top-level (`SDC`) constraints:
```
$ sed -i "s/logical_tile_clb_mode_default__fle_/logical_tile_clb_mode_clb__0\/logical_tile_clb_mode_default__fle_/g" $RTLDIR/SDC_leaf/logical_tile_clb_mode_clb_.sdc # bug in OpenFPGA path naming
$ sed -i "s/fpga_top\///g" $RTLDIR/SDC/*.sdc # all SDC files
$ sed -i "s/logical_tile_clb_mode_clb__0_/logical_tile_clb_mode_clb__0/g" $RTLDIR/SDC/logical_tile_clb_mode_clb_.sdc # bug in OpenFPGA path naming
$ sed -i "s/logical_tile_clb_mode_default__fle_/logical_tile_clb_mode_clb__0\/logical_tile_clb_mode_default__fle_/g" $RTLDIR/SDC/logical_tile_clb_mode_clb_.sdc # bug in OpenFPGA path naming
$ sed -i "s/grid_clb\//grid_clb_1__1_\//g" $RTLDIR/SDC/logical_tile_clb_mode_clb_.sdc # another bug in SDC file naming
$ sed -i "s/grid_clb\//grid_clb_1__1_\//g" $RTLDIR/SDC/logical_tile_clb_mode_default__fle_mode_n1_lut6__ble6.sdc # bug in SDCs
```

> On Sep 25, 2023, at 3:07 PM, Nachiket Kapre ***@***.***> wrote:
> 
> .

tangxifan, 2023-09-25:
@nachiket We have not reworked SDC too much since then. I suggest to double check and see if your changes are still required.

nachiket, 2025-10-28:
We have a sed workaround for now but the patterns are different, there's still some naming bugs in the file hierarchy when you generate SDCs.