lnis-uofu/OpenFPGA #1376: compile error in vtr-verilog-to-routing/abc/abc

anudeepdharavathu, 2023-09-25:
collect2: error: ld returned 1 exit status
make[4]: *** [vtr-verilog-to-routing/abc/CMakeFiles/abc.dir/build.make:98: vtr-verilog-to-routing/abc/abc] Error 1
make[3]: *** [CMakeFiles/Makefile2:1278: vtr-verilog-to-routing/abc/CMakeFiles/abc.dir/all] Error 2
make[2]: *** [Makefile:146: all] Error 2
make[1]: *** [Makefile:76: compile] Error 2
make: *** [Makefile:84: all] Error 2

compilation error after running "make all"

tangxifan, 2023-09-26:
@anudeepdharavathu Can you provide more details? There should be more error messages to indicate where it goes wrong. 

Typically, it is an environment problem. Please refer to our tutorials:

https://github.com/lnis-uofu/OpenFPGA/discussions/937

https://openfpga.readthedocs.io/en/master/tutorials/getting_started/compile/

anudeepdharavathu, 2023-09-26:
I am trying to compile the openFPGA on a ubuntu machine.

1. cloned the git repo
2. installed all the dependencies. 
3. make all

and then I see the above error.
[log.txt](https://github.com/lnis-uofu/OpenFPGA/files/12721746/log.txt)

tangxifan, 2023-09-27:
@anudeepdharavathu According to the log (looks short in my view), it seems that you have tried to compile multiple times. Can you try to start a clean build? 

It is a linker problem. I suggest to check your g++/gcc version and also ensure they are supported by OpenFPGA.
OpenFPGA's build compatibility is continuously checked on Ubuntu machines (fresh machines). See details in : 

https://github.com/lnis-uofu/OpenFPGA/actions/runs/6287186230

anudeepdharavathu, 2023-09-27:
clean build worked.

closed 2023-09-27 by anudeepdharavathu