lnis-uofu/OpenFPGA #1380: Generated spice netlist bug

narutozxp, 2023-09-27:
As i used the command of `write_fabric_spice --file ./SPICE --verbose` to generate spice netlist, the OpenFPGA didn't raise any error. Just like what i expected, transistor.sp was correct, but muxes.sp was not. the two netlists are as follows:
```xml
*********************************************
*	FPGA-SPICE Netlist
*	Description: Transistor wrappers
*	Author: Xifan TANG
*	Organization: University of Utah
*	Date: Wed Sep 27 04:25:36 2023
*********************************************

.subckt lvtpch_wrapper drain gate source bulk L=5.999999786e-08 W=1.199999957e-07
X1 drain gate source bulk lvtpch L=L W=W
.ends
.subckt lvtnch_wrapper drain gate source bulk L=5.999999786e-08 W=1.199999957e-07
X1 drain gate source bulk lvtnch L=L W=W
.ends
```

```xml
************************************
* SPICE module for mux_tree_size12 *
************************************
.subckt mux_tree_size12 in[0] in[1] in[2] in[3] in[4] in[5] in[6] in[7] in[8] in[9]
+                       in[10] in[11] sram[0] sram[1] sram[2] sram[3] sram_inv[0] sram_inv[1] sram_inv[2] sram_inv[3]
+                       out VDD VSS

*********************************
* BEGIN Local short connections *
*********************************
*******************************
* END Local short connections *
*******************************
****************************************
* BEGIN Local output short connections *
****************************************
**************************************
* END Local output short connections *
**************************************

X const1_0_ const1_0_const1 VDD VSS const1

X mux_l1_in_0_ in in[1] sram MUX2_0_Y VDD VSS MUX2

X mux_l1_in_1_ in[2] in[3] sram MUX2_1_Y VDD VSS MUX2

X mux_l1_in_2_ in[4] in[5] sram MUX2_2_Y VDD VSS MUX2

X mux_l1_in_3_ in[6] in[7] sram MUX2_3_Y VDD VSS MUX2

X mux_l1_in_4_ in[8] in[9] sram MUX2_4_Y VDD VSS MUX2

X mux_l2_in_0_ MUX2_0_Y MUX2_1_Y sram[1] MUX2_5_Y VDD VSS MUX2

X mux_l2_in_1_ MUX2_2_Y MUX2_3_Y sram[1] MUX2_6_Y VDD VSS MUX2

X mux_l2_in_2_ MUX2_4_Y in[10] sram[1] MUX2_7_Y VDD VSS MUX2

Omit the following
```
In my opinion, there shouldn't be a space between 'X' and its following character.
How can i solve this bug. Thanks!

tangxifan, 2023-09-27:
@narutozxp FPGA-SPICE has not been ported to OpenFPGA yet. Currently, it is a low priority to us. If you have some strong need, please let us know.

narutozxp, 2023-09-28:
@tangxifan Yes, i just hope to use spice to get the power of FPGA subcircuit. So, is there any method that could be used if FPGA-SPCAE has not been ported to OpenFPGA.
By the way, as I wish to fix the bug, could you give some tips about how to modify the code of OpenFPGA.
Thanks!

narutozxp, 2023-10-27:
@tangxifan Hello, it seems an unnecessary space after the character, "X" , which results incorrect spice file.
https://github.com/lnis-uofu/OpenFPGA/blob/3c50d1cd96def24b71366a5e85b3114d2728e8dd/openfpga/src/fpga_spice/spice_writer_utils.cpp#L260