lnis-uofu/OpenFPGA #1382: Where can I find a file that contains the following timing information?

Chris202305, 2023-09-28:
I'm trying to find the detailed information about the delays of some components or their library files in the 'report_timing.setup.rpt' file provided by OpenFPGA. Where can I find them?
[image]

tangxifan, 2023-10-04:
@Chris202305 Please check the .blif file which is the input to VPR. You can find the net/LUT in the same name.

Chris202305, 2023-10-09:
Can I understand that all the detailed delay information provided by the VPR analysis on the FPGA structure generated by OpenFPGA is described in the architecture file?

tangxifan, 2023-10-09:
Correct.

Chris202305, 2023-10-10:
Okay, with your help, I can now roughly understand the composition of this delay. So, when I use VPR-OpenFPGA to output an FPGA design, VPR will provide a timing analysis report, and then how can I perform Timing Back-Annotation on it?

tangxifan, 2023-10-10:
@Chris202305 Can you clarify the timing back-annotation? I am not sure about the concept in your context.

Chris202305, 2023-10-10:
I'm sorry, I'm not very experienced, and it seems like my words may have caused some confusion. What I meant was, before I started working with OpenFPGA, I was learning to synthesize RTL designs using the DC tool. During my learning process, I discovered that I could generate an SDF file for timing annotation in post-synthesis simulations. My idea is that after VPR completes the placement and routing, it provides a timing report. I'm wondering if it's possible to create a file similar to SDF, where I can annotate the timing information from the architecture file and then perform post-routing simulations to see the results

Chris202305, 2023-10-11:
By the way, can I program an FPGA with the bitstream generated by OpenFPGA using Quartus for testing? I'm currently attempting this and was wondering if you have any experience in this area

tangxifan, 2023-10-14:
@Chris202305 Commercial FPGAs have their in-house configuration protocols and address maps, which are not open to public. If you want to support, you may check some reverse engineering projects (e.g., project X-ray  https://f4pga.readthedocs.io/projects/prjxray/en/latest/) and build your FPGA programmer to support it.

tangxifan, 2023-10-14:
@Chris202305 If you need SDF, you may try the timing-annotated simulation which is supported by VPR: https://docs.verilogtorouting.org/en/latest/tutorials/timing_simulation/