lnis-uofu/OpenFPGA #1385: How do I specify size of the FPGA tiling?

nachiket, 2023-09-29:
I'm trying to generate an FPGA with different tiling configurations i.e. 2x2, 4x4, 8x8, etc.
I see there are two places to edit the VPR size
1. task.conf `openfpga_vpr_device_layout=8x8`
2. OpenFPGA's arch.xml file `<fixed_layout name="8x8" width="6" height="6">`

I can't seem to get OpenFPGA to generate anything I specify in the place of the 8x8. It seems to always generate a 4x4 design. 

I'm using fix_device_route_chan_width_example_script.openfpga shell script.

tangxifan, 2023-10-04:
@nachiket Please check your openfpga shell script. We allow user to defined variables there. 

https://github.com/lnis-uofu/OpenFPGA/blob/36babaaf2d876d563bd2e59133e3ef2aacac207d/openfpga_flow/openfpga_shell_scripts/fix_device_example_script.openfpga#L3

You can see the variable ``${OPENFPGA_VPR_DEVICE_LAYOUT}`` in the openfpga shell script. When using the script, you can define the value of the variable through ``openfpga_vpr_device_layout`` in your task.conf file. 

The layout name, e.g., ``8x8``, must be defined in the vpr arch file.

tangxifan, 2023-10-04:
If the problem insists, please check the openfpgashell.log file and see the options of VPR.