lnis-uofu/OpenFPGA #1388: Need for direct interconnect blocks

msaideroglu, 2023-10-03:
Why do we need direct interconnect blocks in generated netlists? When I synthesize the netlist, the tool put buffers for those direct interconnects and 30% of total delay comes from those buffers. I want to make sure that if the tool puts those buffers just for implementing assign function or the circuit already needs those buffers in order to properly driving all other sub blocks.

tangxifan, 2023-10-04:
@msaideroglu In most of examples of OpenFPGA, the direct interconnection are implemented by ``assign`` clause, with no implication on using buffers. Can you be specific on which blocks are used by buffers? If the synthesis tool adds buffers, please check your synthesis constraints. What we can do is to ensure you a clean netlist where only necessary buffers are defined.

msaideroglu, 2023-10-04:
Yes, the direct interconnection are implemented by assign clause in generated netlist. The synthesis tool adds buffers for those assigns. I could not be sure that if the tool always add buffers for an assign even if there is no need for driving strength, or there is an already need to add buffers so the tool is adding buffers.

closed 2023-10-06 by msaideroglu