lnis-uofu/OpenFPGA #1391: Implementation with SDC constraints

AhmadHouraniah, 2023-10-04:
Hello,

I have a few questions regarding the SDC constraints for the implementation stage.
I am hardening the individual blocks (sb, cb, grid_clbs, and IO pads) separately. Then I will include all the hardened macros in fpga_top and harden the complete design (using OpenLane). However, I'm not sure how I should set and use the SDC constraints.

For routing blocks, I am only using the generated SDC file, e.g. sb_0__0_.sdc for sb_0__0_, while adding a constraint for the programming clock.
For grid_clb blocks, I am using the SDC constraints generated for CLBs and their submodules (also adding clock constraints).
I'm also only applying the SDC constraints generated by `write_sdc_disable_timing_configure_ports `to fpga_top since it doesn't seem relevant when hardening the macros.
1) For IO blocks I see that an SDC constraint is only generated for one of the ports in grid_io_left, as seen below and the reason is not clear to me. Should I apply similar constraints to all IO ports?
```
set_max_delay -from fpga_top/grid_io_left/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/iopad_inpad[0] -to fpga_top/grid_io_left/logical_tile_io_mode_io__0/io_inpad[0] 1.100000022e-10
set_max_delay -from fpga_top/grid_io_left/logical_tile_io_mode_io__0/io_outpad[0] -to fpga_top/grid_io_left/logical_tile_io_mode_io__0/logical_tile_io_mode_physical__iopad_0/iopad_outpad[0] 1.100000022e-10
```
2) Another question is how can I decide on which of the options for `write_pnr_sdc` I should use. I am considering the options listed in the [documentation](https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_sdc_commands/). Currently, I'm not adding any additional options to the `write_pnr_sdc` command, similar to the command used in [SOFA's script](https://github.com/lnis-uofu/SOFA/blob/master/SCRIPT/openfpga_shell_script/skywater_generate_fabric_using_key_example_script.openfpga). Does this enable all the options by default?
3) While looking at the [OpenFPGA architecture file for SOFA](https://github.com/lnis-uofu/SOFA/blob/master/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/openfpga_arch.xml) I see that delays (rise and fall) are in picoseconds. I assume this doesn't significantly affect the generated SDC constraints. Based on this, we are only back annotating the SDC constraints in the VPR architecture. Is this correct?
4) Do you see any issues with my approach to implementation?

Thank you,
Best regards.

AhmadHouraniah, 2024-04-19:
Hello @Lukemagik,
Yes, we did manage to tape out (but we haven't received the chips yet). We also encountered a few errors along the way; however, they are manageable.
We mainly used the approach I described in this issue.

Lukemagik, 2024-04-20:
This is good news @AhmadHouraniah , could I ask you some more specific questions about the entire flow you followed?

1. Did you implement the various individual blocks of the FPGA on silicon and then assemble them together in the FPGA top?
2. If possible, could you provide an example of the config.json file you used on OpenLane to perform synthesis of the individual blocks ?
3. Then, one last question regarding the SDC files, you mentioned that you added constraints on the programming clock, did you manually insert them into the SDC file or in another file? Is there a way to generate it directly on OpenFPGA?

Thank you very much for your availability, I look forward to your response.

AhmadHouraniah, 2024-04-22:
Yes I hardened all the macros separately and them added them as hard macros to the top level. (fpga_top)
The generated SDCs originate at fpga_top. I made small modifications to adjust for the hierarchy and manually inserted the programming clock constraint. OpenFPGA does automatically generate constraints for creating the programming clock, you can copy this into the SDCs for individual macros. The CLB macro should also have a operation clock constraint (as well as prog_clk), so you can adjust the SDCs according to your architecture.

As for the config.json, unfortunately I'm not allowed to share it. But I can share some of the errors we got and how we solved them.

One issue is that the generated SDCs are hierarchical. And I couldn't find a way to set hierarchical SDCs with OpenLane.

During the implementation, the netlist gets flattened. Yosys does not regenerate the SDC constraints so you end up with constraints that don't work with the flattened netlist. You would see these as errors when checking the OpenSTA logs. 
This is only an issue for the CLB macro. If you're generating a tile based architecture I believe you would get this issue for all the tiles as well. 

The delays should be analyzed so that you can back annotate them into VPR architecture. It should be possible to estimate these delays using OpenSTA separately on a non-flattened netlist if you're using a standard cell library approach where you manually select the standard cells to be used. I dealt with this issue by using Synopsys Design Compiler to flatten the netlist and regenerate the SDCs using Design Compiler.

Since we built the FPGA architecture using Standard Cell Libraries, it caused a few errors in the flow since OpenLane is not expecting standard cells in the RTL. Due to this, we added the netlist Verilog models (primitives.v and sky130_fd_sc_hd.v) to the `"VERILOG_FILES_BLACKBOX"` section set `QUIT_ON_LINTER_ERRORS` to 0.

We also had some STA problems for FPGA top due to combinational loops. You can disable STA for the combinational loops using set_disable_timing commands in the SDC (for fpga_top).

Lukemagik, 2024-04-26:
Hello, I'm trying to understand how to generate macros and started from a slightly modified test [example](https://openlane.readthedocs.io/en/latest/tutorials/digital_guide.html). However, I'm having issues with generating the PDN, and I was wondering if you've encountered this error before.

[image]

The error occurs when I try to use the previously created macro mem_1r1w generated with the following config.json file

[image]

and the config.json file that I used for the top module that integrates the macro is:

[image]

Attached is the zip file containing the file that i used for generate the macro and use it.
[problems.zip](https://github.com/lnis-uofu/OpenFPGA/files/15127016/problems.zip)

Thank you in advance. I wouldn't bother you, but I don't understand why it still doesn't work despite following the tutorial.

AhmadHouraniah, 2024-04-26:
Hello,
I'm not sure what's causing the error as I'm not too familiar with setting up the PDN network. I used the caravel user project example as a template.

You can also take a look at the config.json provided by efabless:
[user_proj_example](https://github.com/efabless/caravel_user_project/blob/main/openlane/user_proj_example/config.json
). I used this file as a template for my project.

I think you can get better help for OpenLane from the [open-source silicon slack channel](https://join.skywater.tools).

The community is very active and you should be able to get assistance there.

I hope this helps, and it's no bother at all.

Best regards,
Ahmad.