lnis-uofu/OpenFPGA #1410: Clock and Reset pins Fc is not zero !

WazaAbdulkadir, 2023-10-16:
Hi, 

I have a successful flow. I obtained the netlist, ran a couple of benchmarks on the fabric, and the simulations also look fine. According to the version of VPR I am using, I can't utilize the fc_override tag. As a result, for clock and reset pins, the fc is not set to zero.

Warning (or error) messages are displayed as shown in the following image:

[image]

I declared these ports as global ports in the OpenFPGA architecture.

Since everything seems to be working fine, and we have declared these ports as global inputs, I assume it is not a real issue?

tangxifan, 2023-10-16:
@WazaAbdulkadir Please read the documentation: https://openfpga.readthedocs.io/en/master/manual/arch_lang/annotate_vpr_arch/#cmdoption-arg-tile

If you enable tile annotation, you need these Fc to be 0. Otherwise, your clock may not be connected.

WazaAbdulkadir, 2023-10-17:
I understand it, but the version of VPR I am using doesn't allow me to use the **fc_override** tag. Since I can't set the connectivity for the clock and reset to zero, even though the netlist and simulation seem fine, I'm still getting this warning in the OpenFPGA shell log. I wanted to verify this case.

tangxifan, 2023-10-17:
@WazaAbdulkadir It is hard for me to judge if you are not using the latest master. I do not have bandwidth to offer debugging support. If you can pass design verification, then it should be o.k.

closed 2023-10-17 by WazaAbdulkadir