lnis-uofu/OpenFPGA #1423: How can I open openfpga shell ?

august-35, 2023-11-01:
Hello, I want to try openfpga shell. I've tried the method mentioned at https://github.com/lnis-uofu/OpenFPGA/issues/75#issuecomment-670038931 , but I find there isn't any excutable file under the path /OpenFPGA/openfpga. So with the command /OpenFPGA/openfpga/ ./openfpga -i I couldn't open the shell.
Could you help me?
Thank you !

tangxifan, 2023-11-01:
Hello @august-35 , we have standardized the repo organization especially for the build binaries. Now all the binaries can be found  under the ``OpenFPGA/build/``. For example, the openfpga shell is under ``OpenFPGA/build/openfpga/openfpga``

closed 2023-11-01 by tangxifan

august-35, 2023-11-02:
I've made it, thank you @tangxifan