lnis-uofu/OpenFPGA #1448: Undefined Outputs on Example And2 Benchmark

AllenDBoston, 2023-11-16:
> **Describe the bug**
OpenFPGA fabric verilog configured with a bitstream has undefined outputs. This is true even for the and2 basic example provided in the template tasks/fabric_verification_template. 

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Run-task /openfpga_flow /tasks/template_tasks/fabric_verification_template
> 2. Simulate and2_top_formal_verification_random_tb testbench module using Modelsim

> **Expected behavior**
> With inputs a = 1, b = 1. It is expected that the and2 output(c) is 1. Even for an incorrect bitstream the output should be defined as (1 or 0). 

> **Screenshots**

[image]

tangxifan, 2023-11-16:
@AllenDBoston Can you check if the same issue can be reproduced using iVerilog?

chaitalisathe, 2024-04-10:
@AllenDBoston I am facing similar issue, outputs from FPGA are undefined (X). Is this issue resolved? 
@tangxifan I tried using iverilog, but my benchmark has 36  inputs and 7  outputs, 160 total gates and 62 LUTs, the simulation froze in between. so I am using modelsim, with modelsim I faced issue for "default nettyp none". For time being I have commented out that line. I am able to run the simulation with modelsim now. But I am getting undefined output.
[image]

Could you please point out the issue here which causing all outputs to be undefined.

Thank you,
Chaitali