lnis-uofu/OpenFPGA #1455: Issue in run_fpga_flow.py file parsing

ArslanNustian116, 2023-11-21:
I am following OpenFPGA documentation at this link
https://openfpga.readthedocs.io/en/master/manual/openfpga_flow/run_fpga_flow/

The section describes the bare minimum usage of OpenFPGA
**open_fpga_flow.py <architecture_file> <benchmark_files> --top_module <top_module_name>**

Using this template I ran this command after source OpenFPGA shell file and setting environment variable
**python3 open_fpga_flow.py k6_N10_40nm.xml bnn.v --top_module bnn**

This command produces the error as follows

Traceback (most recent call last):
  File "/home/arslan/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 1060, in <module>
    main()
  File "/home/arslan/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 337, in main
    validate_command_line_arguments()
  File "/home/arslan/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 478, in validate_command_line_arguments
    args.openfpga_arch_file = os.path.abspath(args.openfpga_arch_file)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen posixpath>", line 399, in abspath
TypeError: expected str, bytes or os.PathLike object, not NoneType

tangxifan, 2023-12-05:
@ArslanNustian116 My suggestion is to use run_fpga_task.py all the time.

ArslanNustian116, 2023-12-11:
Thanks @tangxifan 

May be some one can update the document section.

closed 2023-12-11 by ArslanNustian116