lnis-uofu/OpenFPGA #1471: Support LUTRAM mode in FPGA fabric

chungshien, 2023-12-04:
We would like to support LUTRAM mode, where CLB can be configured to dual-port RAM. 

This feaure uniquely exposes the structure of config bits to the customer, we need to enhance OpenFPGA so that customer's view align with OpenFPGA's.

Few changes are needed.

**Change 1**
For QL Memory Bank Flatten BL/WL configuration mode: fix the WL size for CLB tile to be 32bits. Existing OpenFPGA will try to keep BL and WL in a square shape if possible (with BL might be one bit bigger than WL in some cases)

**Change 2**
(I am middle to understand other changes, will update more)

**Describe the solution you'd like**
** Change 1**
1. In architecture XML, introduce "num_wl" attirbute when we declare "pb_type" element
2. This attribute is optional, only special case like this need to specify "num_wl" attirbute, in this case set to 32
3. Eventually this attribute value be trasnferred in new variable "num_wl" introduced in t_physcial_tile_type class. By default the num_wl is 0 (hence the XML attribute is optional):
> If it is 0, OpenFPGA will maintain square shape BL vs WL if possible (original code)
> If it is non-0 (possible number), OpenFPGA will use it as hardcoded size as WL size
4. During build_physical_tile_module(), OpenFPGA will determine BL and WL size based on what had been desscribe in (3)
5. Only physcal tile will be affected, other module building code (routing, grid etc) will not be impacted

chungshien, 2023-12-05:
@tangxifan 

FYI, this is something we planned to do. 

I created the issue so that you can have early review. 

Please comment if you have any concern.

chungshien, 2023-12-05:
@alaindargelas

tangxifan, 2024-01-12:
@chungshien @alaindargelas

I believe this is a good starting point to support LUTRAM.
It is very reasonable to allow users to customize the size of BL/WL per block. However, I do not suggest to add them to the VPR architecture XML, which is designed for a logical/abstract -level representation.
Instead, it should appear in OpenFPGA's architecture description language.
A proper location to add the support is the fabric key file.
For example,

```
<fabric_key>
  <module name="grid_clb" num_wls="32">
```

tangxifan, 2024-01-12:
@chungshien @alaindargelas
My vision is that the fabric key will offer these detailed customization in future. We will keep add more syntax to allow designers to customize any corner of their FPGA. The OpenFPGA's architecture file provides a high-level but rather general description.

chungshien, 2024-01-13:
Hi @tangxifan 

Thanks for reviewing. Any other place to put this info? 

For us, we no longer use fabric_key.xml

So if possible, I would like to avoid fabric_key.xml, but if you really think this is the ONLY BEST way to do so, then I will look into it. 

How about openfpga.xml
> We can define it within "configuration_protocol", since this affect configuration bits arrangement. 

[image]

alaindargelas, 2024-02-29:
@ganeshgore for review too.