lnis-uofu/OpenFPGA #1488: Additional inputs for Fracturable LUTs to increase logic capacity

mustafaarslan0, 2023-12-12:
**Is your feature request related to a problem? Please describe.**
Currently the fracturable LUTs in the OpenFPGA has input size equal to LUT-size (for example, frac-LUT6 has 6 inputs). This is disadvantage when VPR try to use two LUT5 with 5 common inputs. It is decreasing logic capacity.

**Describe the solution you'd like**
As Intel did, we can add 2 more inputs and use it when two LUT5 mode is selected. Then LUT5's use 2 common inputs and 3 independent inputs.
I think this can be done with adding muxes (like OR2 using as tri-state) to select inputs of LUT mux. When two LUT5 mode is selected, additional inputs will be used.

[image]

ducminhnguyen123, 2024-01-04:
If we do like you said above, the execution time of VPR can be reduced a lot?

ducminhnguyen123, 2024-01-04:
I think the biggest problem of OpenFPGA is the time. I implement a design on OpenFPGA and it took me 3 days just for VPR step.

mustafaarslan0, 2024-01-05:
@ducminhnguyen123 

> If we do like you said above, the execution time of VPR can be reduced a lot?

I think it is not directly related with the execution time of VPR. 

> I think the biggest problem of OpenFPGA is the time. I implement a design on OpenFPGA and it took me 3 days just for VPR step.

It is true if design and FPGA is big, but I do not think it is the biggest problem.

mustafaarslan0, 2024-01-15:
@tangxifan It would be very nice to see your opinions.

tangxifan, 2024-01-15:
@mustafaarslan0 You can definitely try this. In the FPL'19 paper, we have tried something similar already. 

https://ieeexplore.ieee.org/abstract/document/8892171

Associated example: https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/depopulate_crossbar/config/task.conf

If you need a very special LUT which OpenFPGA cannot auto-generate, my suggestion is to write your own LUT netlist. However, just make sure that your LUT structure can be supported by OpenFPGA.

mustafaarslan0, 2024-01-16:
@tangxifan In that paper, there is another LUT4 and two additional inputs are used by this LUT4. I think there is no need to add another LUT. Additional inputs should be input for LUT6. 

When I change the LUT input port size to 8, OpenFPGA gives error. Does OpenFPGA decide the LUT size by looking to the input size? (6 input -> LUT6, 8 input -> LUT8) If so we cant define 8-input LUT6 and cant add custom LUT6 netlist that has 8 inputs.

Another method, we may define only LUT4 in OpenFPGA. Then bind LUT6, LUT5 modes to LUT4 physical blocks. Is this possible?

Also I didn't find where is the operating modes which use 8 inputs in the associated example. Paper and example task are not related.